Matchboxscope/Matchboxscope #1: Writing logfiles to SD card

beniroquai, 2022-04-22:
@ethanjli the Anglerfish survived the water but stopped at one point with no reason. Power was available, also replugging didn't resolve the issue. After bringing it to the lab and downloading the images it continued its operation magically..
For debugging it would be helpful to write the serial output to a logfile that gets stored on the SD card. Do you have any experience how to do that? e.g. `Serial.write(SDCard)` :-)

ethanjli, 2022-04-22:
There's probably a library that can take care of this, I can look into it!

beniroquai, 2022-04-22:
That would be great! I stumbled upon some libraries, but I leave this for
you for now ;-)

Message ID: ***@***.***>
>