Matchboxscope/Matchboxscope #7: [FEATURE] Compile code as `.bin` using github actions

beniroquai, 2022-09-13:
Since we have OTA and don'T want to overload the user with difficult library setup in Arduino IDE it would be great to build the binaries and provide it as releases. 
See here: https://github.com/openUC2/UC2-REST/actions

@ethanjli do you have any experience on how to move the artifact to a potential release?

beniroquai, 2022-09-13:
Copy-pasted code from UC2-REST here
https://github.com/beniroquai/Matchboxscope/blob/master/.github/workflows/compileBinary.yml

Trying to compile now. Needs tweaks.

ethanjli, 2022-09-13:
I'm not familiar with methods for uploading binaries to the ESP32, but I'd normally expect it to require running something in the command line, e.g. `esptool.py`. There may be some GUIs available, e.g. a Google search returned http://esphome.github.io/esp-web-tools/

beniroquai, 2022-10-06:
Works now. BIN's will be available in the build artifacts.

closed 2022-10-06 by beniroquai