Matchboxscope/Matchboxscope #12: [FEATURE] Upload images to external repo/GDrive

beniroquai, 2022-09-24:
It would be wonderful to have the images (if connected to the internet) directly uploaded to an external data sink like Gdrive, Dropbox, Github, or alike. I have added a super basic script that uses a GoogleScript to push an image to Gdrive:
https://github.com/beniroquai/Matchboxscope/blob/master/ESP32/main/main.ino#L412 

I think it would be neat to have an additional button to upload it from the GUI. Alternatively, since google is not really open-source, we could think about pushing content to an github repository. Perhaps https://staticman.net/ is the way to go? 

Do you have any experience/opnion @frauzufall, @ethanjli ? :-)

ethanjli, 2022-09-24:
Generally users will need to bring their own API key - I'd suggest implementing only one or two places to upload to where the process for provisioning an API key is relatively straightforward and easy to understand, to illustrate that this is a capability of the project. And then you can encourage people to send you requests to implement any other storage services as upload destinations, once they need it 🙂

beniroquai, 2022-09-24:
Indeed. I have created a GDrive for the anglerfish. With publicly visible
folder/script. This seemed straightforward to me.

Alternatively we could use the REST API. By GitHub I guess, or?

beniroquai, 2023-03-08:
Hey @ethanjli, Since @Vsaggiomo is producing millions of these devices now, I think it would be nice to have a send-to-github-gallery button. 
I created a simple python script 

```py
import requests
import base64

url = 'https://api.github.com/repos/matchboxscope/matchboxscope-gallery/contents/image.png'
headers = {
    'Accept': 'application/[vnd.github+json](http://vnd.github+json/)',
    'Authorization': 'token TOKEN',
}

# Get the current version of the file to update
response = [requests.get(url](http://requests.get%28url/), headers=headers)
[response.raise_for_status()](http://response.raise_for_status%28%29/)
current_file = [response.json()](http://response.json%28%29/)

# Create the updated file content
new_content = 'my new file content'
encoded_content = [base64.b64encode(new_content.encode(](http://base64.b64encode%28new_content.encode%28/)'utf-8')).decode('utf-8')

# Update the file with the new content
data = {
    'message': 'my commit message',
    'committer': {
        'name': 'matchboxscope-bot',
        'email': 'matchboxscope@[gmail.com](http://gmail.com/)'
    },
    'content': encoded_content,
    'sha': current_file['sha']
}

response = [requests.put(url](http://requests.put%28url/), headers=headers, json=data)
[response.raise_for_status()](http://response.raise_for_status%28%29/)

print([response.json())](http://response.json%28%29%29/)
```

and converted that into esp-code:

```cpp

#include <WiFi.h>

const char* ssid = "Blynk";
const char* password = "12345678";

const char* host = "api.github.com";
const int httpsPort = 443;

void setup() {
  Serial.begin(115200);
  delay(1000);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected to WiFi");

  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  client.setCACert(root_ca);

  Serial.print("Connecting to ");
  Serial.println(host);

  if (!client.connect(host, httpsPort)) {
    Serial.println("Connection failed");
    return;
  }

  String data = "{\"message\":\"my commit message\",\"committer\":{\"name\":\"matchboxscope-bot\",\"email\":\"matchboxscope@gmail.com\"},\"content\":\"bXkgbmV3IGZpbGUgY29udGVudHM=\"}";

  String url = "/repos/matchboxscope/matchboxscope-gallery/contents/image.png";
  String headers = "PUT " + url + " HTTP/1.1\r\n" +
                   "Host: " + host + "\r\n" +
                   "User-Agent: ESP32\r\n" +
                   "Accept: application/vnd.github+json\r\n" +
                   "Authorization: token TOKEN\r\n" +
                   "Content-Type: application/json\r\n" +
                   "Content-Length: " + String(data.length()) + "\r\n" +
                   "Connection: close\r\n\r\n";

  client.print(headers);
  client.print(data);

  Serial.println("Request sent");

  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      break;
    }
  }

  while (client.available()) {
    String response = client.readStringUntil('\n');
    Serial.println(response);
  }

  Serial.println("Request complete");
}

void loop() {

}
```

How can we make sure the Token remains invisible? Github won't allow us to expose that to the general public, right?

What this should do essentially is using the matchboxscope-bot account to write to the gallery.

ethanjli, 2023-03-08:
If you need to distribute the API token on each device, then the only way to keep it secret is by distributing only a binary. Then if the users of the device want to modify the Arduino code running on their device, they'll need to provide their own API token. If such a situation would be unacceptable, then you'll need to find an API route which does not require a secret token.

If you need public uploads of files which get shown on a gallery, it may be better to use a public API to upload to somewhere else (e.g. maybe GDrive or Dropbox) where a secret API token is unnecessary, and then use GitHub Actions to automatically pull images from the upload folder and regenerate the gallery website on a regular schedule (e.g. daily or hourly).

beniroquai, 2023-03-08:
Thanks! It's always so fascinating to read your answers. You are an expert for everything! :D 
I guess for now - if it works at all - I like the binary aspect. The CLI build process could eventually read the repo-secret and compile that into the binary I guess (?). 
I'll have a look into the dropbox and TinaCMS. 

For now, the github API throws errors.. or at least the image is not uploaded (plain text works though).

The message from the LOG:

```
 [  6809][D][ping.cpp:233] ping_recv(): Request timeout for icmp_seq 3

[  6809][I][ping.cpp:344] ping_start(): 3 packets transmitted, 0 packets received, 100.0% packet loss

[  6813][I][ping.cpp:349] ping_start(): round-trip min/avg/max/stddev = 1000000000.000/0.000/0.000/-nan ms

Ping failed -> we are not connected to the internet most likely!
Connecting to api.github.com
[  6833][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 155308
[  6839][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[  6869][V][ssl_client.cpp:149] start_ssl_client(): Seeding the random number generator
[  6870][V][ssl_client.cpp:158] start_ssl_client(): Setting up the SSL/TLS structure...
[  6874][D][ssl_client.cpp:179] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[  6882][V][ssl_client.cpp:257] start_ssl_client(): Setting hostname for TLS session...
[  6890][V][ssl_client.cpp:272] start_ssl_client(): Performing the SSL/TLS handshake...
[  8143][V][ssl_client.cpp:293] start_ssl_client(): Verifying peer X.509 certificate...
[  8144][V][ssl_client.cpp:301] start_ssl_client(): Certificate verified.
[  8147][V][ssl_client.cpp:316] start_ssl_client(): Free internal heap after TLS 115132
LED : 255
Send a captured image to Github.
Headers:
PUT /repos/matchboxscope/matchboxscope-gallery/contents/1481765933.jpg HTTP/1.1
Host: api.github.com
User-Agent: ESP32
Accept: application/vnd.github+json
Authorization: token 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
Content-Type: application/json
Content-Length: 57536
Connection: close

[  8616][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 298 bytes...
[  8625][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 57536 bytes...
Request sent
<!DOCTYPE html>

<!--

Hello future GitHubber! I bet you're here to remove those nasty inline styles,

DRY up these templates and make 'em nice and re-usable, right?

Please, don't. https://github.com/styleguide/templates/2.0

-->

<html>

  <head>

    <title>Unicorn! &middot; GitHub</title>

    <style type="text/css" media="screen">

      body {

        background-color: #f1f1f1;

        margin: 0;

        font-family: "Helvetica Neue", Helvetica, Arial, sans-serif;

[... 53 lines cut ...]

        font-size: 14px;

        margin: 0 10px;

      }

    </style>

  </head>

  <body>

    <div class="container">

      <p>

        <img width="200" src="data:image/png;base64,

[ 69878][V][ssl_client.cpp:324] stop_ssl_socket(): Cleaning SSL connection.
iVBORw0KGgoAAAANSUhEUgAAAZAAAAGZCAMAAACQbpc2AAADAFBMVEWEBz6FAD6FAD6GAD+MAEGOAEOKAEGOAEOGAD+IAECOAEOOAUOOAEOOAEOOAEOOAEOOAEOOAEOCACyOAEQAAACKAEJpoJ2KADqu0eSKAD2BAD+KAD6AxCNqwoX1Ziawo9LYnGuwhL7aiwaIyYudst6DlrnhcWvCcVvlbE9PvsL2u3uYmdCZum9Rns7clUYsreOYp2nuj37kiZLUfqhvxafqfmrNWEjA22uz1Vu6lsiRAArWhLfgdoONnF36rWrzpz/ajy/LbqR8pdWDVXx8faN6m0+6hE3+1pmHP2mTy1b4k1p/ueFlj7vMfzyBs1v/2GfLpFL5rQD+7teQABj4n0/rpCz6r1Z8apP+4rrfw1iV0JyLUkqQACTDnah5AACtaEt9zcGNhleJbU5lsOFRvE5FuOPS4WLSt7+8ajvXcJpcyNvhztSwUkTGgZz9uDSxRnPZwsnTkqy7jpsPt/AxyvnAeJMAxPnMiqPJYIyzf4/KqrPedaK5bIsQwfTo2t72hUepPWvn6mO8VoXrgbDRaJXle6r75FzZm7S2TnyTNUXv5eiDAA6j0WmKyWv51liRAACjP0OZNVuwa4L+uQAcw/SiNGH57mIxxPKmU3DgpL2sX3rutc3/vh9gu1D3tNEAtvOSKVAmxfagRmbnrcWZJVaTJET2qsv84eyk04PzcEb71OX0u9OUzHD1nsR+1PhtwFUBvfQ5x/RizvdQy/VZzPWaz3b8uST+wFWDABqKAC6m03uGx2X+xGKBxmL2iWDQjb777/Nt0Pf+xmlCyPT+v032hVv+yXD3lGx2wlr99/hJyvX1gFV20vf1e1DCaqr6xdz+wlzJe7T2jmWFACT3i7v+0IP9u0R8xF7Oh7r+zHn/+WLHc7CQGUP3vtePEUT0i7qHADORIUyEAC2BAACf0Hr3j73zhLX7x96RGkrMgbePFEiRDUeOA0P4wNiVAEaPBkTzh7f9yeCPCkSLADmRAESYAEf+/f6CAET4xdz4wtnzibiKAEH/9GH///+OAEP4w9pHeYEoAAAAFXRSTlP9+PLorFm9i97MnGlJOnoNKxwBAwB644ahAACClklEQVR42uydf0xUZ77/rQLyS4TBk4Yh2XTdxLrurr3qVcyuRdzvt7CSRYSrLMS7S/R6WZqy9Mem2dt+myLG2E4VOnXipLWYGnXStTXx/lOUq/QSgwgxWpEgFccTkE0cYZyJZ2YcyeDj9/15nvODmQG1OpR1t+8znBlmEPW85vPz+ZwzM+b9A8o0b85c7J5K/SMCMc2bG/cDkL8jZc5LmTnvByB/N8qcl/lM4rynVf+QQGYHEp5WA/nHA2LKnBfv9yQDy9OpfzQg4JHKmCftByB/H8o0gYc/5In/AchjKTMz1uYxL83vlxhL/wHIY/HALbY4MhICjDH/M09tGQIg04gjIw0HzhQ7HJlpIY+kuCVPwlNrIAAynUQSslNxFwvXB6xzU2YF/JJTYu7A0xvTpxnInIAnLoMi8RPIZOKRKD1tJnDITrPVEfLP+QHI4/WcMmf5A+ZUHNMngQFlpCRIwOF2SOb1FZJ/5twfgDx2TR3ye2bj0eOwEDAy01MSQoEAAw7ZbKnoX292xD29PKYTCL27Uz1MYoE42Mp3+4PEQsBInsUCMA5Jdihmtn5n/1hl0PX0ViEAMr0571yzn7FQIAk+5pFRmMTDjNS0OA6DSYrT4ZTMLL+xf2xsp9s/4yn2WNNpIXMyKM/ySCDiSXz4MRQuijQ3PT4tLjHkGRkZdkIOJwuZg9b1jf3gQR7r6W1kTScQpFjPxKOwDgAIUzyzQOQRYMydkzI7YZbidDkcTkUKQWazOeitXF9hBw5SZciZOi9T6GnkMp0uKymQkDHHAxzMLwcSM+eZJqWhskhOMivDDscIC5ohyW+x1lTm56+v2DkGGlz9FSEnynRdgPKUlezTCSTe4zGnZfsVtmWT34kljAmJgAbBSEuYyQIjLngnkLBUgkJFY6PdbgcKaExTfz48Fnxaanx8fGp6hlY1fvcU7p8SSEY288NAnP6XtlpgI7MncFq8Ao+Pm+kPBEacTsBglfkVjWQRYSB0Ho2S7ElOfoZ5SMycmJySnvndmWQCyT8fECghEGKMyf4tW/fLTKGeeXR/ak7yMx5kUk7UfCFrPrwTBzGZYCDMjx/nAm7UJ7OSU+cK5/XoPOaiWP3nA5IpIjqAbNq69SXcmTPCDwPhSPBTBU7GYcmvsE/KQosgQSdjjKcJsux0yrLiBh/PTJ2J6dHsIwHW+s8IZI6faUC2boHTQoUd5q0y4uDSQgwVeKiyYueDYWgpFv1CpGBMQg4mSRJz4puRgH9mcnyGFuVND6lWM+MCqdMHxDRtiQjGp2YSEYVZtkIWOC3jONCDlFCA4Yg6zVJ+xdhDcYgahCmoEXkGVlNZWWN1MkqLJRlQPNkJaZif06mYJsvn5swKZGdMs8uanozdhCAiERD [... cut ...] 7eVDIp4oxXqG0rMXYsFemj16Gdi9Sg9ek8gM+N1EiZiZbh9cz4AGNqj/GQv1m9VBtKYsE/OtMCNhDzOUuDQoBh0hcsw5q1DKl+PlRYgb786zT27sFtqYXF+nsZEYFnSfKEYWa4HHBX0uEj0Wbu7nZ36yn0uMfyeRGTsczx+UCKzPIhItYrggQHpTt94CzR57MJcIkRu5Ptvvmc8llKV+4eL6UTA3DtvmgUiIq13eL4uJCTLzgZnft5hQEmSlKpCkRiI6PFGMNpip3so6EpobOERFubO1DgUZPJT4uHqEMniqmu94EAFgThdARNMZ3cSRus8N1p530MEFYOqHC5WFTEst0ogmzcn3XiEYn6SBDGkf7DBml+Ul1NaDiwCxaDR1Qs0AzQOLruw
Request complete
LED : 0
[ 72924][V][ssl_client.cpp:324] stop_ssl_socket(): Cleaning SSL connection.
```

ethanjli, 2023-03-09:
What HTTP response status code did you receive as a result of your HTTP request to upload the image?

> The CLI build process could eventually read the repo-secret and compile that into the binary I guess (?).

That's a great idea for getting started! Be aware that it'll be possible for someone to look inside the binary to read your API key.

beniroquai, 2023-03-12:
Some update to send JPEGs via ArduinoJson:

```cpp
#include <WiFi.h>
#include <HTTPClient.h>
#include <ArduinoJson.h>
#include "Base64.h"

const char* ssid = "Blynk";
const char* password = "12345678";

const char* token = "TOKEN";
const char* owner = "matchboxscope";
const char* repo = "matchboxscope-gallery";
const char* path = "image1234.jpg";
/* This works in MAC
curl -L \
  -X PUT \
  -H "Accept: application/vnd.github+json" \
  -H "Authorization: Bearer TOKEN"\
  -H "X-GitHub-Api-Version: 2022-11-28" \
  https://api.github.com/repos/matchboxscope/matchboxscope-gallery/contents/test2.jpg \
  -d '{"message":"my commit message","committer":{"name":"Monalisa Octocat","email":"octocat@github.com"},"content":"iVBORw0KGgoAAAANSUhEUgAAAOEAAADhCAMAAAAJbSJIAAAAulBMVEX////gRCL7+/vv7+/q6urgQR3eMADgQBv30cvdJwD98u/fPBXskIDfOhD53dj//fziVDrkWTrhSyn1xbvytKnulYLiTy/gRSXpiHj87Oj++PbpfWfpfWv98e/eNAD76ufnbFTvno7lYknkY0/yr6P308zrhW/75N/tj3vtl4nwp5n0wbjkW0DyrqD0wrnum43ncVvmd2jlaVblcF/siXPrgWnlXj7kVjPcGQDmb17gSS7kXEXvp530vK9oV1uaAAAMbUlEQVR4nO2dDXuiOBeGZXeTIDGgiCCgEalIBbVMa8e+7fj//9bLp1/FtiMZLLt5rmtmbIqBm4RzTk4SptXi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi+s/r77//apKuIfyH+U37ZuKEzRcnbL44YfPFCZsvTvi7shS29VUXY0Ip0CSmFVYXY0IZkMWYaY2VxZoQAtKxmFZZVawJ2wKgC6ZVVhVjQq8tCICsmNZZUX+AUABoM2FaayWx76WJaK/PtNoqYm5pUkIBq9p3sTesCWlGKCASfBOvwZiwXxDGJvWbeA22hHKEhQPi4FuEN0wJJfcAmDyMi+8QpDIljOgxYOw1AoaVXyuGhJJNhFMBoDGr/WqxIzTn54BxP+2KrKq/WswI5RF+Bxi7/s7NfQYjQmUdkdgHEgLO+ql68+CGDaH/oGMB9KazoXHeT51be0UWhNbsDqMYZqq0fAedIcJp9RNUUnVCJXRA+giC7mBqv3sSAbrxOKMqoRW6pDAxgJASa0Pu2VzptapK2G+/9xHnjXjb4K0qoZhZz8SM4uQPBu8QyW2fxMq9dJEggvl0Easzt+9UQlJjA/YdFt3dtBErWxqzg7CAAklRLEsyfVnUhjqNXYe7t6qge1OfWN2WjmdbirsbTc4HEspYDnqUzuXnfRx+027KwB8q/gBDCJb2NJwUJQsYjRU176hoeMtRFJuYZqzZWwLbbbwceBmNNpVaXm5m8Y9bxjXMIm9L3gX38y2kUcaY/HWfIeJHn9FJrhHbLIbyjAVCgsJ25nkpdCezPMlvii3h+GdsQAEc5kMm08H/NsIwsy14kbWiNU0bEd/9a3qptMx8IFB3WUGYET6ZDE/yu2JJaNmFB8SjjElESRSHR/8KW3qSiQJGmBZ5IFoCAXWYneMKsSOUnaNcIvmVlnlT+QcS1JsmFVkRKv2342EUvk/NqSVZc9D4uDSVudJPBr/Izs2nNQRNH1ukkof4ND+DnJxQGQJ820E+E8Lw7l0W0c2dvDVE5Jb+ngmhFLy+S88cCOfEqXyGSqpO6Nv4PIMYEy697LfSCN44J1yZUOyVpaL2hJNHt/I1VlPlTJRQmmvb91L5zat+kZVUkVDulc3HxIROTtifNjurP3EvZEtRlHsL8ZbDilSVCKU5LQc8ePzxrZuwGmHw3ojmwvNbDphOVIVQfrtM2PkW6zASVSGcXgQUSHCSQJS81SZynOh5KtaeWKxAaI4uE+LDEgXL1GyBQkoSQboN/HofzQqEcnSRkDwWsajkdXRKjg7EMWOtM4p/ghBjO1+CYXkLlZ57TExGXo19lX0vxWQ5y12Ft1iSsmOIW+MilEqWpmzGlyyDfI7GX7yVxOTZXVjWF8tVITSdM4ePKHS1SWZIpJV7iS+5Eb3a+mklj29GkIBs7AuSWWB1I+Z2Ugndk/4JUDJDTA4zxLC2gf+1hEqQJtB2dteILx8Yr649LfBaVt85si8AI6N7Z8+HG9vpCjkkUOsa+V9NOMgeJUVez4JgtfMOHmAs2vQQkCOsPi5COYZXLMuUZyM9a1tc156Fqwm1dlRmLRQ/HKEDHyDdTmi2xrK4Cxad+8VsHU4dNSW/q8nxX/0cihA763NzEUdnkXCUlcJoKFpSfzZ01fQ5JAT1RnYvOQDoNWU3rib0iIC7neNmNMWYBBwtNwHUCcdeMNLx0SIUhA01+21NmfCrCWUdxDaku9FEf2L64m46TMwIOsoqYjwdy4MlKFlik6LW9CBeTThJV5MAjIVX/VWNcc4XCyFdFG3jsk9EnXpc4tWE0qKwJwCAklYCPwc9eJ4oPiEc1mNqrvf44ScL2lRcnqRqTBt+NML/kr79c9iyHj5blfixQE0rpSrEpf3eB0/Z5zJq2qlQgVAafvygneks9gbdmgZQVcYWofrVRkQEQvTTeX6OfhKY3Rf0yA7iQ1UaPS2+0ojxwAn1nlde7hvMVbY5qrZ500qE4+hTY4OIepfE3keSOmnT1zXpVm3eQu5+iBhHPG/D3bvM2niOBeTWlVOsOPckLi931HgQFQVeGUioCrC2/d6V5w975ZEZwEQYhn552CIuyba2tHDlOWBv/m6zU5L2JW4gX5y6EJc1Tn1Xn8e3QofC41EhhnB7L44/iDpD8lBfZp/JahNv00MkF1LdwWcpbXlV4549RmuiLG83vd9s7gcr79tMHObi7xhqvjhh88UJmy9O2HxxwuaLEzZfnDCTojCbYyiv6cIJWJz2hNAyx3vtB0HKRF5rM02UD2PasWnuR7CKdPRDWhB/+ajAin8svij54m6mrWXz9MLHvqiVFct9bbbzJhUHyyeEsmPvFeWnM2cOoBBCqo7WxblWtr2fdJgMbOdkHfC4Y9tHCfuZbReTTF6nR+KqKHBmR/k3pT/X0xMIzu5oYGz1R0ZSTJbTasuMTwjFl/gK2ple8oU/Q0ooAQaI/1GDHHFnQKM4bQhh93TEPoLU3t9384m2861r/WVcBzaEuD46OnSRnR7/jNJictjkZmk6IUQwECUwqrTS75SwLfScXMvsxgcEgNFO9MSZQxDI80dShIq5sbFDyOy0d8kILPfr1sQlwhmNpGLcfViL4m4OENwUB/g6Jm4giqJmI9SeFcVeNy7WRHHdiSGHDAnxUJ7kyor+h4WBlBAo/hwTJ2+tFS42NokU6Odrf+4w/lV8ngFiZ5+CNnL7Sdsq40AA7aIRpxA7aVJVMYcAdfObZf2i2E3Xx1mBil6rrC46JzxbvSy/oW5RIqoAFv3FQEJ6u5Upgb/ODd4O4qfiwAWlecNvEXjIj5QjtM8I63i/2rbfRWre+JJNioSj6aBKyzY+IYxP2i0+yz0Ai5upQfIjaQVpi7fvZsksDEh+1eIWq3mpANQCRRpiMsg/w8PCGvMOFRuITReTol4b4Srd9EpCC4H01u/auCT1uSE4e8eXMiOwuP/GgdA6IqTHhLiYNjVdQnNCJX4+q2yzvZKwNaDphgMBl72pdALRMj3StzEpKvwK4VEbInJjQp8kz0wI8agkd2/ZhGjJMyca9L5o4s8JrfUuLLae9ne73BbdilBZULKwIvwaltW6JiQyEzsIwd5vfE5YrlsRtjwB6SsdR6Uxlf8TJYdOdHyId74N4Wl8cplwvMFAFcCgVSYroCSO1XaQLPZ37FrC1q3aMPb1giAYF65OFJBqKg6C633R9W0IWBKC5fM80XOxffAyoTkkAnIuVCstYo/tE+wcvvCbhIocRD97idRqL2U4j0txFnm/5AbiA8KWRoX27lK9oUqiFTxEb58QKn0xUxELSSsD5kJsCXU3VbEf4gPCyYgI8OK6Jj8OzntoedQ8HxJK7XQ5EQH5LVN2GJPlKNWWKSEaiXIiT/qUMFmbCHoXZ7IXWAAnzvJDwvELpTQevBiFx3cwmYsTM9YkYvocft3SjOcYIMEodYdpXV1wutD5Q0JLC3fhrLuPaSZtoB/i0tsQygh1HQFdXtk0wqcvxPjY0iSJmqOozWujYtx7s5jmHtLA0/HlPQXTuJMe//xbcWm/jewbR22TF6CLrWd8+YXWTSd8JmQ4bnkQ/7i0MqHhhCYGRhKuuES/5BJZEd7I0kxp5glEeHHX9hcJy8f4R4TWU62ExfDtFWWXa7lYv7DW95xQAGqRSZNO2lAtYlff3ROKB0Jv+Uf9ofyG9CKQ6gtCkYkKUJ5rUwJCF+Xm9Jxwi9A+ExWH5AXtFuMiG9nfotf8dskvwM3upjnCTGOad33uiYB5RjBxEYmy35oOwnkb+PB9NjHTOaEWW98sPpA6SHgpEnSrdpFgm9gIvOalEwLILiskCP9RQg0jom5W2sqOgyqUs2gq6uYNN7YJKX99yTlhq4cxdae71b1OweEt35OkOAq01cYgoF2ESNZDfIy90gYqpZ5D2BG+0HNCZQAwge1kAgF38/87xvpB4X47iEgpKnUYD4TaJwXyHcbJWKFNsHG0I0h8y4ohIcZhk4kfxaXxaelroPTbzLKJ/rATnreHoo1cXTXU7mOneOOD/NDp7KHMoDMsnTrpLzpnIw9/+rTUBVV3hyf/zY78EHVVw4iLjx2P//AYn7ZrxxdkDS4O0r6g0/nD0vm6sddfh2tRPkyonc72XZz7e18+Efvhuu+d9xNfXIdh/2zFreL348LJxcv6qvgccPPFCZsvTth8ccLmixM2X5yw+eKEzRcnbL44YfPFCZuv/wThXw3SFYRXfYmLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi4uLi6vJ+j8CXRUnlwc4sgAAAABJRU5ErkJggg=="}'
  */
  
const char* message = "my commit message";
const char* committer_name = "matchboxscope-bot";
const char* committer_email = "matchboxscope@gmail.com";
[... cut ...]
  http.addHeader("Content-Type", "application/json");
  int http_code = http.sendRequest("PUT", payload_string);

  Serial.println(payload);
  Serial.println(payload_string);
  
  // Check the HTTP status code
  if (http_code == 200) {
    String response = http.getString();
    Serial.println(response);
  } else {
    Serial.println("Error: " + http.errorToString(http_code));
    Serial.println(http.getString());
  }

  http.end();

  // Wait for some time before sending the next request
  delay(5000);
}
```

beniroquai, 2023-03-12:
@ethanjli it seems to be working. Finally. A lot of fiddling around and I guess the Base64 library is the limiting factor here. 
I do:

1. Capture an image
2. Convert the buffer into a base64 encoded String
3. Add all metadata to a `arduinojson` object
4. Add the content
5. Upload that to the Github API using the bot's credentials and the token 
6. The final image appears here: https://github.com/Matchboxscope/matchboxscope-gallery/blob/main/image_1481765933.jpg

## Problem:
- the image appears not idea ;-) 
- Do you have any idea what could cause the issue? It looks surprisingly similar to the serial thingy...

[image]

@KillerInk maybe you have an idea? ;-)

Relevant code:
https://github.com/Matchboxscope/matchboxscope-fw/blob/main/main/main.ino#L1235

## Observation:

It works with previously generated base64 images as can be seen here:
https://github.com/Matchboxscope/matchboxscope-gallery/blob/main/image1270216262.jpg

KillerInk, 2023-03-12:
looks like the image get corrupted^^
theory one:
not sure why you are converting the bytearray to base64 for http.
the protocol do support it.
What you need for the request is:

```
Content-Type: image/jpeg
Content-Length: [length in bytes of the image]
```
and then directly read out the framebuffer to http request.
maybe the base64 cause it.

theory two:
maybe there is race condition happening with autofocus and imagecapture
the frame buffer gets filled with new data while it get readout.
esp_camera_fb_get seems not to block writes to the framebuffer

the second theory is more likely

https://github.com/espressif/esp32-camera/blob/master/README.md#jpeg-http-capture

beniroquai, 2023-03-15:
Autofocus? I understand trainstation ^^
Could we do a deep copy from the buffer before it gets overwritten then?

beniroquai, 2023-03-15:
@KillerInk - I would need to use this: https://github.com/espressif/esp32-camera/blob/master/README.md?plain=1#L233
for the PUT request for github, right? I have mixed feelings about github accepting it. It's really picky. After some trial and error, I got the code above to work. Many other snippets failed with weird HTTP responses from Github.