Matchboxscope/Matchboxscope #19: ESP32 fails to open SD card in write mode

beniroquai, 2023-01-30:
It happened again. The Anglerfish is constantly waking up and takes images until this happens:

```
Message appended
Appending to file: /debug.txt
Message appended
LED : 255
LED : 0
LENS Value: 255
LENS Value: 0
Appending to file: /debug.txt
Message appended
Appending to file: /debug.txt
Message appended
In timelapse mode.
Appending to file: /debug.txt
Message appended
/anglerfish_3624_Z_0
LENS Value: 0
LED : 255
LED : 0
LENS Value: 0
Saved /anglerfish_3624_Z_0.jpg
/anglerfish_3624_Z_25
LENS Value: 25
LED : 255
LED : 0
LENS Value: 0
Saved /anglerfish_3624_Z_25.jpg
/anglerfish_3624_Z_50
LENS Value: 50
LED : 255
LED : 0
LENS Value: 0
Saved /anglerfish_3624_Z_50.jpg
/anglerfish_3624_Z_75
LENS Value: 75
LED : 255
LED : 0
LENS Value: 0
Failed to open file in writing mode
/anglerfish_3624_Z_100
LENS Value: 100
LED : 255
LED : 0
LENS Value: 0
Failed to open file in writing mode
/anglerfish_3624_Z_125
LENS Value: 125
LED : 255
```

After this bootloops and no written images unfortunately. The error is happening [here](https://github.com/Matchboxscope/Matchboxscope/blob/master/ESP32/main/app_httpd.ino#L108) 

@ethanjli any guess what could cause this issue? A formatting of the micro sd card with the official SD Card formatted resolves the issue. I suspect some bad blocks may get written on the SD card or the index is ruined? I can read the card on a windows computer afterwards. Written files seem to be intakt.

beniroquai, 2023-01-30:
What's really interesting ist that the SD card gets properly mounted and some files (e.g. `Debug.txt`) get written:

```

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

Wakeup was not caused by deep sleep: 0
E (884) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first
Stored date:
Jan 30 2023 08:15:02
Compiled date:
Jan 30 2023 08:15:02
First run? no
exposureTime : 220
gain : 0
timelapseInterval : 180
Camera init success!Camera is working
SD Card Mounted
2
Listing directory: /
  DIR : .Spotlight-V100
  DIR : .fseventsd
  FILE: debug.txt  SIZE: 64
  FILE: anglerfish_3609_Z_0.jpg  SIZE: 5010
  FILE: anglerfish_3609_Z_25.jpg  SIZE: 5010
  FILE: anglerfish_3609_Z_50.jpg  SIZE: 5045
  FILE: anglerfish_3609_Z_75.jpg  SIZE: 5045
  FILE: anglerfish_3609_Z_100.jpg  SIZE: 5088
  FILE: anglerfish_3609_Z_125.jpg  SIZE: 5081
  FILE: anglerfish_3609_Z_150.jpg  SIZE: 5208
  FILE: anglerfish_3609_Z_175.jpg  SIZE: 5432
  FILE: anglerfish_3609_Z_200.jpg  SIZE: 5614
  FILE: anglerfish_3609_Z_225.jpg  SIZE: 5778
  FILE: anglerfish_3609_Z_250.jpg  SIZE: 5958
  FILE: anglerfish_3610_Z_0.jpg  SIZE: 5140
  FILE: anglerfish_3610_Z_25.jpg  SIZE: 5162
  FILE: anglerfish_3610_Z_50.jpg  SIZE: 5161
  FILE: anglerfish_3610_Z_75.jpg  SIZE: 5186
  FILE: anglerfish_3610_Z_100.jpg  SIZE: 5186
  FILE: anglerfish_3610_Z_125.jpg  SIZE: 5212
  FILE: anglerfish_3610_Z_150.jpg  SIZE: 5280
  FILE: anglerfish_3610_Z_175.jpg  SIZE: 5450
  FILE: anglerfish_3610_Z_200.jpg  SIZE: 5629
  FILE: anglerfish_3610_Z_225.jpg  SIZE: 5766
  FILE: anglerfish_3610_Z_250.jpg  SIZE: 5949
  FILE: anglerfish_3611_Z_0.jpg  SIZE: 5156
  FILE: anglerfish_3611_Z_25.jpg  SIZE: 5171
  FILE: anglerfish_3611_Z_50.jpg  SIZE: 5171
  FILE: anglerfish_3611_Z_75.jpg  SIZE: 5171
  FILE: anglerfish_3611_Z_100.jpg  SIZE: 5171
  FILE: anglerfish_3611_Z_125.jpg  SIZE: 5182
  FILE: anglerfish_3611_Z_150.jpg  SIZE: 5266
  FILE: anglerfish_3611_Z_175.jpg  SIZE: 5444
  FILE: anglerfish_3611_Z_200.jpg  SIZE: 5595
  FILE: anglerfish_3611_Z_225.jpg  SIZE: 5702
  FILE: anglerfish_3611_Z_250.jpg  SIZE: 5925
  FILE: anglerfish_3612_Z_0.jpg  SIZE: 5146
  FILE: anglerfish_3612_Z_25.jpg  SIZE: 5140
  FILE: anglerfish_3612_Z_50.jpg  SIZE: 5166
  FILE: anglerfish_3612_Z_75.jpg  SIZE: 5166
  FILE: anglerfish_3612_Z_100.jpg  SIZE: 5166
  FILE: anglerfish_3612_Z_125.jpg  SIZE: 5195
  FILE: anglerfish_3612_Z_150.jpg  SIZE: 5245
[... 168 lines cut ...]
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

Wakeup was not caused by deep sleep: 0
E (884) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first
Stored date:
Jan 30 2023 08:15:02
Compiled date:
Jan 30 2023 08:15:02
First run? no
exposureTime : 220
gain : 0
timelapseInterval : 180
Camera init success!Camera is working
SD Card Mounted
2
```

Interplay between camera + SD card?

ethanjli, 2023-01-30:
Hmmm, my first guess (without having deeply inspected the logs) is that it's possible some rare fault might have occurred if you're still using one of the SD card pins to control the camera LED.

beniroquai, 2023-01-30:
Indeed, I also suspect a hardware problem. Do you happen to know if one could skip-broken sectors on the SD card and/or write to another random address in case of failure? 
What's weird though is that the txt file gets written - but indeed it gets overwritten.. So most likely an SD card problem :/

ethanjli, 2023-01-30:
> Do you happen to know if one could skip-broken sectors on the SD card and/or write to another random address in case of failure?

SD cards should automatically skip broken sectors when writing - I believe this feature is called wear leveling.

> What's weird though is that the txt file gets written - but indeed it gets overwritten..

What do you mean when you say that the txt file gets overwritten - and is that behavior you would expect to occur if everything were working correctly?

beniroquai, 2023-02-01:
Definitely a broken SD card problem. Entered a new one and now it's working for 100 scans wihtout any problem. 

Regarding the TXT file: I wanted to have a debug option in case the device is not connected ot the USB Serial. Therefore I established this function: https://github.com/Matchboxscope/Matchboxscope/blob/master/ESP32/main/main.ino#L472

In case of the above error, this file gets (over?)written without problems. However, new JPEG files won't get written to the (damaged?) SD card.

closed 2023-02-06 by beniroquai