Matchboxscope/Matchboxscope #22: [FEATURE] External LED control

arteys, 2023-03-24:
I'm working on adding an external LED, it would be nice to be able to control it from the software. You can connect it to the 3v output, but then it will be constantly on and will quickly drain the battery.

beniroquai, 2023-03-25:
Hey @arteys! Thanks for reaching out. I have added a slider for a PWM channel (GPIO12):
[image]

It's a general purpose pin that can be used for the motor, a voice-coil-driven lens or an LED (eventually with a transistor). 
I also used a neopixel array once - of interest? In combination with the SD card, there are not many pin configurations that will work with SD card + GPIOs .. 

the current firmware can be found here: 
https://github.com/Matchboxscope/matchboxscope-simplecamera/tree/matchboxscope

If wanted, I can add binaries to the firmware uploader @Vsaggiomo ?

arteys, 2023-03-25:
Thank you very much! Uploader is that mega-convenient web page that allows you to download software to the controller? Then yes please.

Vsaggiomo, 2023-03-25:
@beniroquai if it's only the PWM, it makes sense to integrate it with the bin file. It shouldn't break anything (last famous words :D )

arteys, 2023-04-01:
I'm sorry it took so long, but I finally got to the new version. And.. it doesn't compile. What about the last words?

I'm probably stupid and doing something wrong, however.

> `c:/users/modern/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Modern\AppData\Local\Temp\arduino\sketches\294A0BD34909B1E58B055BB9CD5D549A\sketch\storage.cpp.o:(.literal._Z9loadPrefsRN2fs2FSE+0x8c): undefined reference to `jsonExtract(String, String)'
c:/users/modern/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Modern\AppData\Local\Temp\arduino\sketches\294A0BD34909B1E58B055BB9CD5D549A\sketch\storage.cpp.o: in function `loadPrefs(fs::FS&)':
C:\Users\Modern\Desktop\Python\matchboxscope-simplecamera\main/storage.cpp:95: undefined reference to `jsonExtract(String, String)'
c:/users/modern/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Modern\Desktop\Python\matchboxscope-simplecamera\main/storage.cpp:98: undefined reference to `jsonExtract(String, String)'
c:/users/modern/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Modern\Desktop\Python\matchboxscope-simplecamera\main/storage.cpp:99: undefined reference to `jsonExtract(String, String)'
c:/users/modern/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Modern\Desktop\Python\matchboxscope-simplecamera\main/storage.cpp:100: undefined reference to `jsonExtract(String, String)'
c:/users/modern/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Modern\AppData\Local\Temp\arduino\sketches\294A0BD34909B1E58B055BB9CD5D549A\sketch\storage.cpp.o:C:\Users\Modern\Desktop\Python\matchboxscope-simplecamera\main/storage.cpp:102: more undefined references to `jsonExtract(String, String)' follow
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1`

beniroquai, 2023-04-04:
Ok, I compiled the latest version and pushed it to the flashing tool: https://matchboxscope.github.io/firmware/FLASH.html
It's an early stage software, but should at least give you the PWM on Pin 12 :)

arteys, 2023-04-05:
Thank you, but... After flashing controller just screaming in agony from serial port.
The old firmware is installed and works normally

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
invalid header: 0x3025203a
ets Jul 29 2019 12:21:46
```

beniroquai, 2023-04-05:
yep. Saw that last night directly. I'm diving into the odds of partition schemes :D 
Do you have the chance to use the latest platform.io code and try installing it manually for now? 
Link: https://github.com/Matchboxscope/matchboxscope-simplecamera/tree/matchboxscope

arteys, 2023-04-06:
That's where I started. It doesn't compile, and I don't really understand what it's missing.

beniroquai, 2023-04-06:
Fixed latest binary. Should work now. Was an issue with the ESP-IDF
#classy..

> Message ID: ***@***.***>
>

arteys, 2023-04-07:
Whoo! The firmware is installed and it seems to work. The next problem is WiFi. I send {"ssid":"SSID_NAME","password":"SSID-PASSWORD"} via serial, but I don't see any response.

beniroquai, 2023-04-07:
Fixed. Was an issue if no known wifi was found. Should work now. You can
also use you phone's hotspot at "omniscope", "omniscope" ;-)

> Message ID: ***@***.***>
>

arteys, 2023-04-08:
Step by step .. It connected to the wifi, but instead of the familiar page, I see an almost empty page, with links leading to the void. Is something broken somewhere else?

[image]
[image]

beniroquai, 2023-04-12:
Hey @arteys, this is indeed strange. Can you open the serial and copy/paste the output? It's indeed very strange. Something went wrong with uploading the code - but that's however very unlikely since the webpage is hardcoded into c-code. Hmm.. More likely that the server interrupted the communication. I have never seen this. Which board are you using?I have updated the FW. Would you mind trying it again? Does the non-advanced version work? 

@Vsaggiomo yours worked right?

Vsaggiomo, 2023-04-13:
yes, the latest version of the advanced firmware is working (I tried on the omniscope/omniscope one). the streaming also worked. Used the console to check which page it opened (172.20.10.6): 

[ 10182][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 10
[ 10182][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 1 - SCAN_DONE
10 networks found
  1 : [6E:52:CB:52:CC:DA] omniscope (-54)  -  Known!
Connecting to Wifi Network 0: [6E:52:CB:52:CC:DA] omniscope 
[ 10265][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
..[ 11448][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: omniscope, BSSID: 6e:52:cb:52:cc:da, Channel: 6, Auth: WPA2_PSK
[ 11450][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 11543][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:172.20.10.6
[ 11544][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 11547][D][WiFiGeneric.cpp:996] _eventCallback(): STA IP: 172.20.10.6, MASK: 255.255.255.240, GW: 172.20.10.1
.Client connection succeeded
IP address: 172.20.10.6
Netmask   : 255.255.255.240
Gateway   : 172.20.10.1
Setting httpURL
Setting up OTA

No OTA password has been set! (insecure)

[ 13788][I][ArduinoOTA.cpp:141] begin(): OTA server at: Matchboxscope.local:3232
Added HTTP service to MDNS server
Time functions disabled
Starting web server on port: '80'
Starting stream server on port: '81'

Camera Ready!
Use 'http://172.20.10.6/' to connect
Stream viewer available at 'http://172.20.10.6:81/view'
Raw stream URL is 'http://172.20.10.6:81/'
Camera debug data is disabled (send 'd' for status dump, or any other char to enable debug)
Portal page requested
Stream viewer requested
Stream requested
Stream failed, code = 45062 : ESP_ERR_HTTPD_RESP_SEND
Stream ended

Dump requested via Web

Name: Matchboxscope
Firmware: Apr 12 2023 @ 10:13:33 (base: 4.0)
Sketch Size: 1342320 (total: 1966080, 68.0% used)
MD5: 3d772c8e216bab657b2dffed9eb5ec94
ESP sdk: v4.4.3
WiFi Ssid: omniscope
WiFi Rssi: -39
WiFi BSSID: 6E:52:CB:52:CC:DA
WiFi IP address: 172.20.10.6
WiFi Netmask: 255.255.255.240
WiFi Gateway: 172.20.10.1
WiFi Http port: 80, Stream port: 81
WiFi MAC: 78:E3:6D:19:B1:DC
System up: 0:00:01:23 (d:h:m:s)
Active streams: 0, Previous streams: 1, Images captured: 0
CPU Freq: 240 MHz, Xclk Freq: 8 MHz
MCU temperature : 53 C, 129 F  (approximate)
Heap: 263900, free: 99788, min free: 73692, max block: 67572
Psram: 4192107, free: 4000107, min free: 4000107, max block: 3997684
Spiffs: 113201, used: 0
Preferences file: 
/esp32cam-preferences.json not found, nothing to dump.

beniroquai, 2023-04-16:
Hmm... The output on my end looks as follows:

```
[   486][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Stored date:
Apr 16 2023 21:01:16
Compiled date:
Apr 16 2023 23:01:32
First run? yes
Setting Timelapse Interval to: 4294967295
Setting up timelapse: 0
Getting timelapse value 2: 0
Is Acquire Stack is: 1
Getting timelapse value: 0

====
esp32-cam-webserver: Matchboxscope
Code Built: Apr 16 2023 @ 23:01:32
Base Release: 4.0

Starting internal SPIFFS filesystem
Listing SPIFFS directory: /
  FILE: esp32cam-preferences.json       SIZE: 352
Camera init succeeded
OV2640 camera module detected
First light is 0
Old 0x0 value is0
Getting first frame at
997
Got first frame at
1294
SD Card Mounted
3
Creating task saveCapturedImageGithubTask
Tasks created...
Creating task handleSerial
You can enter your wifi password and ssid via a json string e.g.{"ssid":"SSID_NAME","password":"SSID-PASSWORD"}
Lamp: 0%, pwm = 0
[  1374][D][main.ino:917] setup(): PWM pin: 12
Lamp: 20%, pwm = 54
Lamp: 0%, pwm = 0
Loading preferences from file /esp32cam-preferences.json
{"lamp":100,"autolamp":1,"framesize":9,"quality":12,"xclk":8,"min_frame_time":0,"brightness":-1,"contrast":0,"saturation":0,"special_effect":0,"wb_mode":0,"awb":0,"awb_gain":1,"aec":0,"aec2":0,"ae_level":0,"aec_value":720,"agc":0,"agc_gain":4,"gainceiling":1,"bpc":0,"wpc"Serial commandsource /opt/anaconda3/envs/imswitch39/bin/activate /opt/anaconda3/envs/imswitch39

:1,"raw_gma":1,"lenc":0,"vflip":0,"hmirror":0,"dcw":1,"colorbar":0,"rotate":"0"}
internal files System found and mounted
In livestreaming mode. Connecting to pre-set Wifi
Starting WiFi
Known external SSIDs:  'omniscope' 'Blynk' 'XXXX'
MAC address: 78:E3:6D:0B:0C:A8
Scanning local Wifi Networks
[  3093][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  3186][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[  3187][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[ 10196][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 26
[ 10197][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 1 - SCAN_DONE
26 networks found
Connecting to Wifi Network 2: [78:DD:12:D4:2B:AE] XXXX 
[ 10321][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 10397][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: XXXX, BSSID: 78:dd:12:d4:2b:ae, Channel: 6, Auth: WPA2_PSK
[ 10398][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
.....[ 12955][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.2.202
[ 12956][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 12959][D][WiFiGeneric.cpp:996] _eventCallback(): STA IP: 192.168.2.202, MASK: 255.255.255.0, GW: 192.168.2.1
.Client connection succeeded
IP address: 192.168.2.202
Netmask   : 255.255.255.0
Gateway   : 192.168.2.1
Setting httpURL
Setting up OTA

No OTA password has been set! (insecure)

[ 15355][I][ArduinoOTA.cpp:141] begin(): OTA server at: Matchboxscope.local:3232
Added HTTP service to MDNS server
Time functions disabled
Starting web server on port: '80'
Starting stream server on port: '81'

Camera Ready!
Use 'http://192.168.2.202/' to connect
Stream viewer available at 'http://192.168.2.202:81/view'
Raw stream URL is 'http://192.168.2.202:81/'
Camera debug data is disabled (send 'd' for status dump, or any other char to enable debug)
```

From your output it seems your browser aims to reach the portal - weird! 

Can you try entering the URL from the scope manually? 

http://192.168.2.202/?view=full

Where I assume your IP would be a little different ;)

beniroquai, 2023-04-16:
You could also try to get the output of the browser's logs through the developer tools:

[image]

Perhaps we can diagnose errors there. I cannot reproduce the error on my end right now unfortunately.

arteys, 2023-04-17:
I'm beginning to think that I'm under some kind of curse...

Adding "?view=full" somehow helped, I finally saw a normal page. "Get still" was working, but the stream wasn't. For some reason I decided to stick an SD card in...
But when connected to the computer, the cloud of magic smoke left it, after which, of course, it stopped working at all. 

Well, I've got another one! I'll report back.

PS. The only idea is that it was a very bad SD card and this problem hopefully has nothing to do with the software.  

Translated with www.DeepL.com/Translator (free version)

arteys, 2023-04-17:
[image]

Controller number two. Streaming is still not working. Could this problem be due to some strange effect of my router?

Can't the default firmware behaviour be to work in access point mode?

beniroquai, 2023-04-17:
Sorry for the bad SD card and burning ESP.. Not sure how many cams and MCUs I ruined :/ 
Can you try the following URL: 192.168.0.106:81 please? This should give the RAW streaming output. As long as you see images it should not be a camera issue. Would you mind to copy-paste the Serial printout again? Perhaps we can identify some errors on this thread.

arteys, 2023-04-19:
When i connect to 192.168.0.106:81 i saw just a static picture. And serial output was:

Added HTTP service to MDNS server
Time functions disabled
Starting web server on port: '80'
Starting stream server on port: '81'

Camera Ready!
Use 'Undefined' to connect
Stream viewer available at 'Undefinedview'
Raw stream URL is 'Undefined'
Camera debug data is disabled (send 'd' for status dump, or any other char to enable debug)
Portal page requested
Stream requested
Stream failed, code = 45062 : ESP_ERR_HTTPD_RESP_SEND
Stream ended
Stream requested
Stream failed, code = 45062 : ESP_ERR_HTTPD_RESP_SEND
Stream ended
Stream requested
Full index page requested
Stream failed, code = 45062 : ESP_ERR_HTTPD_RESP_SEND
Stream ended
Stream requested
Stream failed, code = 45062 : ESP_ERR_HTTPD_RESP_SEND
Stream ended

beniroquai, 2023-04-20:
Hm. That's indeed weird. I have added a function that erases the camera settings on first boot after flashing. The settings remain in the flash memory and if they are weirdly formulated the camera stream won't start. You can either press the restore and reboot button in the GUI or reflash the newest firmware I just pushed (which should land on the website in ~5 minutes.) :)

arteys, 2023-04-20:
And now i have an infinite camera scream in seral.
I checked - the old firmware still works fine, so it's not the controller or the camera.
It's also not my router - I switched on my laptop in access point mode and saw the same picture.

[ 26747][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26754][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26761][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26768][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26775][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26783][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26790][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26797][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26804][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26811][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26819][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26826][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26833][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26840][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26847][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26855][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26862][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26869][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND
[ 26876][V][Preferences.cpp:400] getUInt(): nvs_get_u32 fail: cameraTLI NOT_FOUND

beniroquai, 2023-04-20:
Interesting, bit should still work. No?
This is a camera-unrelated warning.

arteys, 2023-04-21:
Nope :(

[image]

beniroquai, 2023-04-21:
That's a tough one :D 
Would you mind to copy/paste the entire logs from boot to the moment you hit the start stream button? Perhaps the chrome-debugging information can help (developer tools => Console)

arteys, 2023-04-23:
So... I changed the settings of my access point on my laptop to omniscope - omniscope and... everything worked. I haven't tested the PWM yet, but at least the stream works.

Apparently it's some kind of problem related to the non-default wi-fi. 

Basically, I can live with that now. However, maybe that's why you can't reproduce my error?

arteys, 2023-04-23:
OK. The error appears the first time you connect to a new network. If the controller already remembers the network and reboots itself, the stream works fine. Maybe I'm sending {"ssid": "SSID", "password": "PASSWORD"} at some maximum unfortunate moment?

Now I can't reproduce it myself, because instead of page with unworkable streamer I see page with inability to access the camera. 

Suggestion: maybe better the standard firmware installed via https://matchboxscope.github.io/firmware/FLASH.html will work in hotspot mode like the classic version? That seems easier to me, though less secure.

And besides, the router at my institute doesn't catch inside the incubator, I've checked, so I'll still have to configure it from my laptop.

beniroquai, 2023-04-23:
Thanks for reporting it! 
There is also "improv" which could help. There is an example here: https://jnthas.github.io/improv-wifi-demo/
I could try implementing it as an alternative to the current way of entering the json string? Or: We send a json string that says SSID: AP, which will turn the wifi to AP mode with matchboxscope SSID?

beniroquai, 2023-05-23:
Hey, this looks really gorgeous! #neverhadexpected that! <3 
This is for sure a special case. well..not really. So you want to have the led the same value every time lapse point, right? Hmh. Makes sense. Will work on it.

beniroquai, 2023-05-23:
Have fixed it. Would you mind trying again (unfortunately disassembling the ESP for flashing I guess?) Please test before remounting. ;-)

arteys, 2023-05-24:
Thank you! I'll put up a time lapse with cells as best I can, but probably not this week. Slight renovation in the laboratory.

arteys, 2023-05-24:
Hm, https://matchboxscope.github.io/firmware/FLASH.html tell me that "ESP32-CamServerAdvanced 2023-04-05?". Is this correct  version of firmware?

beniroquai, 2023-05-25:
Yep. It's the same. Date is hardcoded. Forgot to change. Sorry ;)

arteys, 2023-07-03:
So, its me again. My lab was successfully renovated and I was about to make timelapses with cells. However... The camera refuses to save images to the SD card.

PS. If you encounter an error "Failed to execute 'open' on 'serialport': failed to open serial port." then the controller drivers are broke down in Windows 11. The instructions from here helped me: https://github.com/esphome/issues/issues/4525  Working driver version: (https://k2me.tistory.com/145)

I am attaching log of my current error:

rst:0xc (SW_CPU_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[     7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   232][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Starting internal SPIFFS filesystem
Listing SPIFFS directory: /
  FILE: esp32cam-preferences.json	SIZE: 567
Camera init succeeded
OV2640 camera module detected
[   993][D][main.ino:622] StartCamera(): Capturing dummy frame 0
Mean intensity: 126
[  1349][D][main.ino:622] StartCamera(): Capturing dummy frame 1
Mean intensity: 117
Stored date:
Jun 16 2023 06:08:53
Compiled date:
Jun 16 2023 06:08:53
First run? no
[  1466][D][storage.cpp:429] getAcquireStack(): Reading get AcquireStack to0
SD Card Mounted
3
[  1544][D][main.ino:841] setup(): LED pin: 4
Lamp: 0%, pwm = 0
[  1560][D][main.ino:849] setup(): PWM pin: 12
Lamp: 20%, pwm = 54
Lamp: 0%, pwm = 0
internal files System found and mounted
[  1969][D][main.ino:1012] initAnglerfish(): Anglerfishmode is: 1
In timelapse anglerfish mode.
Lamp: 0%, pwm = 0
Updating /esp32cam-preferences.json
[  2024][D][storage.cpp:107] writeJsonToSSpiffs(): Opening file for writing
[  2030][D][storage.cpp:112] writeJsonToSSpiffs(): Serializing JSON to String
[  2031][D][storage.cpp:118] writeJsonToSSpiffs(): Writing JSON to file
[  2054][D][storage.cpp:429] getAcquireStack(): Reading get AcquireStack to0
[  2055][E][vfs_api.cpp:204] mkdir(): File system is not mounted
Current: 270%, pwm = 270
[  2066][D][main.ino:1059] initAnglerfish(): Anglerfish: Acquire Exposure Series
Resetting Camera Sensor Settings...
Current: 270%, pwm = 270
Capture Requested for SD card save
Current: 0%, pwm = 0
[  2618][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sdcard/2/data_Jun_16_2023_060852_timelapse_image_anglerfish_2_z270_texp_1.jpg) failed
Failed to open file in writing mode
Resetting Camera Sensor Settings...
Current: 270%, pwm = 270
Capture Requested for SD card save
Current: 0%, pwm = 0
[  3717][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sdcard/2/data_Jun_16_2023_060852_timelapse_image_anglerfish_2_z270_texp_5.jpg) failed
Failed to open file in writing mode
Resetting Camera Sensor Settings...
[  7783][E][main.ino:1003] loadAnglerfishCamSettings(): Camera frame error
[ 11784][E][main.ino:1003] loadAnglerfishCamSettings(): Camera frame error
Current: 270%, pwm = 270
Capture Requested for SD card save
CAPTURE: failed to acquire frame
Resetting Camera Sensor Settings...
[ 27924][E][main.ino:1003] loadAnglerfishCamSettings(): Camera frame error
[ 31924][E][main.ino:1003] loadAnglerfishCamSettings(): Camera frame error
Current: 270%, pwm = 270
Capture Requested for SD card save
CAPTURE: failed to acquire frame
Resetting Camera Sensor Settings...
[ 48064][E][main.ino:1003] loadAnglerfishCamSettings(): Camera frame error
[ 52064][E][main.ino:1003] loadAnglerfishCamSettings(): Camera frame error
Current: 270%, pwm = 270
Capture Requested for SD card save
CAPTURE: failed to acquire frame
Resetting Camera Sensor Settings...

beniroquai, 2023-07-06:
Hey @arteys, thanks for reaching out again! I was able to successfully take a series of images lately in the incubtor - but the cells were a bit boring. Not much going on here ;-) 

[image]

I saw these kinda logs and I would guess it's a broken SD card. Perhaps you can try formatting it with the original SD card formater https://www.sdcard.org/downloads/formatter/

Can you read/write from a computer? What'S the capacity? Sometimes SD-HC works sometimes not. 
I have pushed the latest changes to the website. Always worth a check :) .. 

Oh, and now that I'm talking about it: I have replaced the PWM on Pin 12 with the NEOPIXEL to have better illumination - I'll quickly revert this. It's a compiling switch (but you mentioned you are using the online uploader, right? :)
[image]

arteys, 2023-08-06:
Finally got around to making some groovy cell videos. However, when I tried to flash the new controller - it goes into an endless loop consisting of:

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
E (563) esp_core_dump_flash: Core dump �f�� check failed:
Calculated checksum='5ead4cf9'
Image checksum='00000000'
[     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   236][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Starting internal SPIFFS filesystem
E (239) SPIFFS: mount failed, -10025
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid he�ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid he�ets Jul 29 2019 12:21:46

beniroquai, 2023-08-07:
This device is a mystery. In the manuscript I'm writing I claim that the ESP32-cam is nice since it's available everywhere and reproducible due to its similarity everywhere. Seems not to be the case :D 
I used the firmware uploader https://matchboxscope.github.io/firmware/FLASH.html (for ESP32 camera the advanced server) just today and it worked. Which hardware are you using now? Perhaps a device without PSRAM (unlikely..). Seems the firmware is corrupted. Can you flash "Blink" or anything using the Arduino IDE?