Matchboxscope/Matchboxscope #25: Having Timelapse Value alongside the slider

beniroquai, 2023-04-08:
It would be cool to get the value of the slider directly inside the GUI. I failed. Help! :D 
@ethanjli, do you have a moment to screen this line and perhaps come up with a suggestion how to solve it easily? I tried some javascript stuff, where the label is updated on slider change, but instead of doing so the slider shrunk to be not adjustable anymore. Relevant line of (new firmware) code:
https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/matchboxscope/main/index_ov2640.h#L52

and in code:
```html
              <div class="input-group hidden" id="timelapseInterval-group" title="Timelapse Interval value.&#013;&#013;Warning:&#013;Choose a value for capturing images continously. 0 means no interval.">
              <label for="timelapseInterval">Timelapse Interval (s)</label>
              <input type="range" id="timelapseInterval" min="0" max="600" value="0" class="default-action"
                    oninput="document.getElementById('valTimelapse').innerHTML = this.value" />
                    <div <label id="valTimelapse"></label></div>
            </div>
```

Thank you!! :)

beniroquai, 2023-04-08:
Currently it looks likethis:
[image]

The Anglerfish setting is also very funky :D

ethanjli, 2023-04-08:
I made this snippet to look like your non-working code, and my snippet works: https://codepen.io/ethanjli/pen/VwEYxWZ . So I would suggest starting from my snippet and incrementally changing it to look more like your code until you can reproduce the problematic behavior.

I also noticed a syntax error on this line:
```
<div <label id="valTimelapse"></label></div>
```
You are missing a `>` after `<div`.

beniroquai, 2023-04-08:
Thanks a lot!
I think the code I prototyped first also worked, but inside the existing javascript GUI it fails. Also the codepen.io code doesn't show the number, unfortunately. Any guess? :) 
Current [image]

ethanjli, 2023-04-08:
Hmmm. I just updated my CodePen sandbox (still at https://codepen.io/ethanjli/pen/VwEYxWZ) to exactly match your code snippet at https://github.com/Matchboxscope/Matchboxscope/issues/25#issue-1659511499 (except with the `<div` tag syntax error fixed), and it works on CodePen (the number shows up and everything). If the same code isn't working in your HTML file, do you see any Javascript error messages in your web browser's developer console?

beniroquai, 2023-04-12:
Hm. Thanks for the update. I'll try again. Do you think the `class="default-action"` may cause issues?

ethanjli, 2023-04-12:
Good intuition. If you have JavaScript code elsewhere in your page which is modifying the data event handlers of any element with the "default-action" class once the page loads, then it'll definitely overwrite anything you add into the ondata attribute of yoir HTML element here.

I'll take another look at your JavaScript to see if anything like that might be happening.

ethanjli, 2023-04-12:
Update: yes, you do have JavaScript elsewhere which is setting the `onchange` handler for every element with class `default-action`: https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/matchboxscope/main/index_ov2640.h#L622-L627

However, you should also be aware that you may also be relying on other functionality attached to your `default-action` class for this element: https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/matchboxscope/main/index_ov2640.h#L528-L533

My suggestion is that you should avoid setting the event handlers of your HTML elements using element attributes in the element tags themselves (e.g. "onchange=", "oninput=", etc.), and instead set the event handlers all together in your JavaScript script. You're already doing that for other HTML elements at the bottom of the file: https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/matchboxscope/main/index_ov2640.h#L682-L685

So you should just add more code blocks in that general area to set the `input` event handler for your element of id `timelapseInterval`. This way you can keep the default functionality for `default-action` and then you can subsequently override it for certain elements as needed.

beniroquai, 2023-04-12:
Aha, indeed that makes sense. Will add that to the code and write it to the board tomorrow. Still running the timelapse. Looks good so far! :) Thanks a lot @ethanjli for the insanely detailed answer!! :)

closed 2023-04-13 by beniroquai