Matchboxscope/Matchboxscope #28: Only one stack

Jordvl, 2023-05-04:
Thanks for fixing my last problem.., im kinda new to github :)
Anyways, I have a new problem... Which is when I setup the fish, then take it out of the pc and plug it into a power supply. I can still log into it and set it into angler fish mode. However once this happens it only takes one stack of images and never again... Obviously id like it to keep taking images for a few days if possible...
Heres my settings, i dont have any logs as it either deletes logs or i have to reset the device to even look into it...
Thanks for helping :D
[image]
[image]
[image]

beniroquai, 2023-05-10:
Hey @Jordvl, 

sorry for taking longer. I manage to get the timelapse with non-focussing working. it creates an exposure series. Did you try this? I have uploaded the latest version. Sshould be ready for flashing online.

How Wifi works: It tries to connect to the default SSID: `Blynk`, `12345678`. If not available, it creates an AP `Matchboxscope-X`, where `X` is a number between 0...9. It blinks `X`-times to indicate the number. (Helpful if you have more. It's getting displayed on Serial too. 
In the GUI you can now set SSID/PW, hit save and reboot. Then it connects to your network. Sounds good? 
CC: @ethanjli is this a plausible Wifi-structure?

Output of the timelapse:
```
[    18][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   467][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Starting internal SPIFFS filesystem
Listing SPIFFS directory: /
  FILE: esp32cam-preferences.json       SIZE: 539
Camera init succeeded
OV2640 camera module detected
[   920][D][main.ino:405] StartCamera(): Capturing dummy frame 0
Mean intensity: 136
[  1282][D][main.ino:405] StartCamera(): Capturing dummy frame 1
Mean intensity: 140
[  1382][D][main.ino:405] StartCamera(): Capturing dummy frame 2
Mean intensity: 135
[  1482][D][main.ino:405] StartCamera(): Capturing dummy frame 3
Mean intensity: 140
[  1582][D][main.ino:405] StartCamera(): Capturing dummy frame 4
Mean intensity: 135
Stored date:
May 10 2023 20:37:17
compiled_date:
May 10 2023 20:37:17
First run? no
Updating /esp32cam-preferences.json
[  1698][D][storage.cpp:107] writeJsonToSSpiffs(): Opening file for writing
[  1708][D][storage.cpp:112] writeJsonToSSpiffs(): Serializing JSON to String
[  1710][D][storage.cpp:118] writeJsonToSSpiffs(): Writing JSON to file
SD Card Mounted
3
Lamp: 0%, pwm = 0
[  1838][D][main.ino:580] setup(): PWM pin: 12
Lamp: 20%, pwm = 54
Lamp: 0%, pwm = 0
internal files System found and mounted
In timelapse anglerfish mode.
Lamp: 255%, pwm = 5672
Updating /esp32cam-preferences.json
[  2193][D][storage.cpp:107] writeJsonToSSpiffs(): Opening file for writing
[  2198][D][storage.cpp:112] writeJsonToSSpiffs(): Serializing JSON to String
[  2199][D][storage.cpp:118] writeJsonToSSpiffs(): Writing JSON to file
Resetting Camera Sensor Settings...
[  2277][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 0
Mean intensity: 94
[  2402][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 1
Mean intensity: 130
[  2451][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 2
Mean intensity: 94
[  2501][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 3
Mean intensity: 130
[  2551][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 4
Mean intensity: 94
[  2601][D][app_httpd.cpp:1332] saveImage(): Setting PWM Value
Current: 0%, pwm = 0
Capture Requested for SD card save
Lamp: 255%, pwm = 5672
Lamp: 0%, pwm = 0
Saved /data_May_10_2023_203717_timelapse_image_anglerfish_0texp_1
Lamp: 0%, pwm = 0
Resetting Camera Sensor Settings...
[  3150][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 0
Mean intensity: 95
[... 230 lines cut ...]
Lamp: 0%, pwm = 0
Resetting Camera Sensor Settings...
[  6356][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 0
Mean intensity: 94
[  6363][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 1
Mean intensity: 130
[  6452][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 2
Mean intensity: 94
[  6502][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 3
Mean intensity: 130
[  6552][D][main.ino:848] loadAnglerfishCamSettings(): Capturing dummy frame 4
Mean intensity: 94
[  6603][D][app_httpd.cpp:1332] saveImage(): Setting PWM Value
Current: 0%, pwm = 0
Capture Requested for SD card save
Lamp: 255%, pwm = 5672
Lamp: 0%, pwm = 0
Saved /data_May_10_2023_203717_timelapse_image_anglerfish_1texp_500
Lamp: 0%, pwm = 0
Updating /esp32cam-preferences.json
[  7116][D][storage.cpp:107] writeJsonToSSpiffs(): Opening file for writing
[  7121][D][storage.cpp:112] writeJsonToSSpiffs(): Serializing JSON to String
[  7122][D][storage.cpp:118] writeJsonToSSpiffs(): Writing JSON to file
Sleeping for 36 s
```

beniroquai, 2023-05-14:
I guess the SD card may ruin the experiment. Some (very long) logs:
[log.txt](https://github.com/Matchboxscope/Matchboxscope/files/11471666/log.txt)

In short:

```

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[    20][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   468][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Starting internal SPIFFS filesystem
Listing SPIFFS directory: /
  FILE: esp32cam-preferences.json	SIZE: 540
Camera init succeeded
OV2640 camera module detected
[  1233][D][main.ino:416] StartCamera(): Capturing dummy frame 0
Mean intensity: 141
[  1595][D][main.ino:416] StartCamera(): Capturing dummy frame 1
Mean intensity: 144
[  1695][D][main.ino:416] StartCamera(): Capturing dummy frame 2
Mean intensity: 141
[  1795][D][main.ino:416] StartCamera(): Capturing dummy frame 3
Mean intensity: 144
[  1895][D][main.ino:416] StartCamera(): Capturing dummy frame 4
Mean intensity: 141
Stored date:
May 11 2023 01:09:18
compiled_date:
May 11 2023 01:09:18
First run? no
Updating /esp32cam-preferences.json
[  2011][D][storage.cpp:107] writeJsonToSSpiffs(): Opening file for writing
[  2024][D][storage.cpp:112] writeJsonToSSpiffs(): Serializing JSON to String
[  2025][D][storage.cpp:118] writeJsonToSSpiffs(): Writing JSON to file
[  2048][D][storage.cpp:465] getAcquireStack(): Reading get AcquireStack to0
SD Card Mounted
3
Lamp: 0%, pwm = 0
[  2151][D][main.ino:591] setup(): PWM pin: 12
Lamp: 20%, pwm = 54
Lamp: 0%, pwm = 0
internal files System found and mounted
In timelapse anglerfish mode.
Lamp: 255%, pwm = 5672
Updating /esp32cam-preferences.json
[  2600][D][storage.cpp:107] writeJsonToSSpiffs(): Opening file for writing
[  2605][D][storage.cpp:112] writeJsonToSSpiffs(): Serializing JSON to String
[  2606][D][storage.cpp:118] writeJsonToSSpiffs(): Writing JSON to file
[  2630][D][storage.cpp:465] getAcquireStack(): Reading get AcquireStack to0
[  2630][D][main.ino:914] initAnglerfish(): Anglerfish: Acquire Exposure Series
Resetting Camera Sensor Settings...
[  2694][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 0
Mean intensity: 131
[  2766][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 1
Mean intensity: 142
[  2815][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 2
Mean intensity: 132
[  2865][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 3
Mean intensity: 142
[  2915][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 4
[... 37 lines cut ...]
E (24728) diskio_sdmmc: sdmmc_write_blocks failed (263)
[ 24752][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sdcard/data_May_11_2023_011635_timelapse_image_anglerfish_70texp_5.jpg) failed
Failed to open file in writing mode
Resetting Camera Sensor Settings...
[ 24824][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 0
[ 28825][E][main.ino:860] loadAnglerfishCamSettings(): Camera frame error
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d7057  PS      : 0x00060c30  A0      : 0x800d779c  A1      : 0x3ffb26a0  
A2      : 0x00000000  A3      : 0x400d778f  A4      : 0x3f42c48d  A5      : 0x0000035c  
A6      : 0x3f41858c  A7      : 0x00000000  A8      : 0x80127d66  A9      : 0x3ffb2680  
A10     : 0x00000000  A11     : 0x3ffb4a20  A12     : 0x0000003f  A13     : 0x0000004c  
A14     : 0x3f417fbd  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4008ca4d  LEND    : 0x4008ca5d  LCOUNT  : 0xfffffff9  

Backtrace: 0x400d7054:0x3ffb26a0 0x400d7799:0x3ffb26c0 0x400d7a52:0x3ffb26f0 0x400d803e:0x3ffb2760 0x400e6652:0x3ffb2820

ELF file SHA256: 01a4e8db51a3b02e

Rebooting...
ets Jun  8 2016 00:22:57
```

This means it fails to write to the SD card and at one point it won't get mounted and reboots to the normal mode.

beniroquai, 2023-05-14:
Interestingly, it only happens after timestamp 41. Here it crashes:

```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[     7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   456][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Starting internal SPIFFS filesystem
Listing SPIFFS directory: /
  FILE: esp32cam-preferences.json	SIZE: 540
Camera init succeeded
OV2640 camera module detected
[  1259][D][main.ino:416] StartCamera(): Capturing dummy frame 0
Mean intensity: 141
[  1621][D][main.ino:416] StartCamera(): Capturing dummy frame 1
Mean intensity: 144
[  1721][D][main.ino:416] StartCamera(): Capturing dummy frame 2
Mean intensity: 141
[  1821][D][main.ino:416] StartCamera(): Capturing dummy frame 3
Mean intensity: 144
[  1921][D][main.ino:416] StartCamera(): Capturing dummy frame 4
Mean intensity: 141
Stored date:
May 11 2023 01:09:18
compiled_date:
May 11 2023 01:09:18
First run? no
Updating /esp32cam-preferences.json
[  2036][D][storage.cpp:107] writeJsonToSSpiffs(): Opening file for writing
[  2046][D][storage.cpp:112] writeJsonToSSpiffs(): Serializing JSON to String
[  2049][D][storage.cpp:118] writeJsonToSSpiffs(): Writing JSON to file
[  2115][D][storage.cpp:465] getAcquireStack(): Reading get AcquireStack to0
SD Card Mounted
3
Lamp: 0%, pwm = 0
[  2219][D][main.ino:591] setup(): PWM pin: 12
Lamp: 20%, pwm = 54
Lamp: 0%, pwm = 0
internal files System found and mounted
In timelapse anglerfish mode.
Lamp: 255%, pwm = 5672
Updating /esp32cam-preferences.json
[  2668][D][storage.cpp:107] writeJsonToSSpiffs(): Opening file for writing
[  2673][D][storage.cpp:112] writeJsonToSSpiffs(): Serializing JSON to String
[  2674][D][storage.cpp:118] writeJsonToSSpiffs(): Writing JSON to file
[  2698][D][storage.cpp:465] getAcquireStack(): Reading get AcquireStack to0
[  2698][D][main.ino:914] initAnglerfish(): Anglerfish: Acquire Exposure Series
Resetting Camera Sensor Settings...
[  2763][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 0
Mean intensity: 131
[  2840][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 1
Mean intensity: 152
[  2890][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 2
Mean intensity: 132
[  2940][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 3
Mean intensity: 152
[  2990][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 4
Mean intensity: 132
[  3040][D][app_httpd.cpp:1334] saveImage(): Setting PWM Value
Current: 0%, pwm = 0
Capture Requested for SD card save
Lamp: 255%, pwm = 5672
Lamp: 0%, pwm = 0
E (3950) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (3951) diskio_sdmmc: sdmmc_write_blocks failed (263)
[  3957][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sdcard/data_May_11_2023_011635_timelapse_image_anglerfish_41texp_1.jpg) failed
Failed to open file in writing mode
Resetting Camera Sensor Settings...
[  4030][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 0
Mean intensity: 132
[  4037][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 1
Mean intensity: 132
[  4140][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 2
Mean intensity: 132
[  4240][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 3
Mean intensity: 132
[  4340][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 4
Mean intensity: 132
[  4440][D][app_httpd.cpp:1334] saveImage(): Setting PWM Value
Current: 0%, pwm = 0
Capture Requested for SD card save
Lamp: 255%, pwm = 5672
Lamp: 0%, pwm = 0
E (9794) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (9795) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (9795) diskio_sdmmc: sdmmc_write_blocks failed (263)
E (14801) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (14801) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (14802) diskio_sdmmc: sdmmc_write_blocks failed (263)
E (19808) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (19808) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (19809) diskio_sdmmc: sdmmc_write_blocks failed (263)
E (24815) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (24815) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (24816) diskio_sdmmc: sdmmc_write_blocks failed (263)
[ 24827][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sdcard/data_May_11_2023_011635_timelapse_image_anglerfish_41texp_5.jpg) failed
Failed to open file in writing mode
Resetting Camera Sensor Settings...
[ 24900][D][main.ino:857] loadAnglerfishCamSettings(): Capturing dummy frame 0
[ 28901][E][main.ino:860] loadAnglerfishCamSettings(): Camera frame error
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d7057  PS      : 0x00060c30  A0      : 0x800d779c  A1      : 0x3ffb26a0  
A2      : 0x00000000  A3      : 0x400d778f  A4      : 0x3f42c48d  A5      : 0x0000035c  
A6      : 0x3f41858c  A7      : 0x00000000  A8      : 0x80127d66  A9      : 0x3ffb2680  
A10     : 0x00000000  A11     : 0x3ffb4a20  A12     : 0x0000003f  A13     : 0x0000004c  
A14     : 0x3f417fbd  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4008ca4d  LEND    : 0x4008ca5d  LCOUNT  : 0xfffffff9  

Backtrace: 0x400d7054:0x3ffb26a0 0x400d7799:0x3ffb26c0 0x400d7a52:0x3ffb26f0 0x400d803e:0x3ffb2760 0x400e6652:0x3ffb2820
```

It sounds like a corrupted SD card.
@ethanjli do you know any mechanism that probes sectors on the SD card and skips them if necessary?

beniroquai, 2023-05-14:
The last bit however is interesting. The SPIFFS get corrupted, too. The value of "anglerfish mode yes/no" is getting reset and this causes the ESP32 to boot up in normal mode. This is completely weird.. 

@ethanjli I have the feeling that the EEPROM is a bit more "persistent". Would it be ok to store the "anglerfishmode yes/no" into the `preferences" again?

beniroquai, 2023-05-14:
At least the idea of having exposure stacks works nicely :) 
@ethanjli 
[image]

[image]

beniroquai, 2023-05-14:
This seems to be related:
https://github.com/espressif/esp-idf/issues/4773
Even though I'm not 100% sure if it's maybe a broken SD card. 

Idea/suggestion:
Since we are not using the LED/Torch as the illumination source anymore, perhaps we can fully ignore it and use full 4-pin mode for saving files to the SD card and don't touch any rtc-stuff in the configuration? What do you think @ethanjli 

We are changing the LED's pin's function here:
https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/matchboxscope/main/main.ino#L962
and here:
https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/matchboxscope/main/main.ino#L884

Otherwise the LED might be slightly on all the time..

beniroquai, 2023-05-14:
Formatting the SD card solved the issue temporally. Lets see for how long.

beniroquai, 2023-05-14:
One more info @ethanjli the stacktrace from above's error leads to some base64 error. Interesting. 

```

0x400d7054: StartCamera() at /Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/matchboxscope-simplecamera/main/main.ino line 289
0x400d7799: initAnglerfish(bool) at /Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/matchboxscope-simplecamera/main/main.ino line 908
0x400d7a52: initAnglerfish(bool) at /Users/bene/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/WString.h line 359
0x400d803e: setup() at /Users/bene/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/WString.h line 364
0x400e6652: base64_encode_blockend at /Users/bene/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/libb64/cencode.c line 82
PC: 0x400d7057: StartCamera() at /Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/matchboxscope-simplecamera/main/main.ino line 289
EXCVADDR: 0x00000008

Decoding stack results
0x400d7054: StartCamera() at /Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/matchboxscope-simplecamera/main/main.ino line 289
0x400d7799: initAnglerfish(bool) at /Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/matchboxscope-simplecamera/main/main.ino line 908
0x400d7a52: initAnglerfish(bool) at /Users/bene/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/WString.h line 359
0x400d803e: setup() at /Users/bene/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/WString.h line 364
0x400e6652: base64_encode_blockend at /Users/bene/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/libb64/cencode.c line 82
```

ethanjli, 2023-05-14:
> CC: @ethanjli is this a plausible Wifi-structure?
The scheme you described sounds good to me!

> Idea/suggestion:
> Since we are not using the LED/Torch as the illumination source anymore, perhaps we can fully ignore it and use full 4-pin mode for saving files to the SD card and don't touch any rtc-stuff in the configuration? What do you think @ethanjli 

Yes, if you can return the SD card-related pins to only be used for the SD card, that will be a great improvement in making the system's behavior easier to understand and troubleshoot.

> Interestingly, it only happens after timestamp 41.
Are you able to get the crash happening always+only after timestamp 41? i.e. is this a deterministic failure, or a non-deterministic failure? Also, are you able to reproduce the exact same behavior on multiple different SD cards?

> @ethanjli do you know any mechanism that probes sectors on the SD card and skips them if necessary?
I believe you're looking for wear-leveling functionality: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/wear-levelling.html

> @ethanjli I have the feeling that the EEPROM is a bit more "persistent". Would it be ok to store the "anglerfishmode yes/no" into the `preferences" again?

If you need to store that one into EEPROM, that's fine. However, I would encourage you to still:

1. Keep one unified interface for managing the persistence of all settings together, so that it provides a clean abstraction over which specific settings are stored on EEPROM vs. SPIFFS, and which internally handles the saving and loading of settings to EEPROM and SPIFFS
2. Try to identify the root cause of that behavior - if the SPIFFS is getting corrupted, there may be a software bug outside of SPIFFS functionality which is contributing to the buggy behavior (e.g. if you're using the SPIFFS API in a way in which it wasn't intended to be used), and which could cause problems even if you store the anglerfishmode setting in EEPROM.

> the stacktrace from above's error leads to some base64 error.
I'm very surprised to see that it's a base64 _encoding_ error. Can you provide a permalink to line 908 of main.ino?

beniroquai, 2023-05-16:
> > CC: @ethanjli is this a plausible Wifi-structure?
> 
> The scheme you described sounds good to me!

Cool. This seems to be working now. One more thing I tried implementing is "Improv" (https://www.improv-wifi.com/) which helps you set up Wifi through the online-flashing tool. I followed this demo and implemented relevant code in the matchboxscope firmware. However, the timings seem to be off. If you try setting up the wifi on the device after flashing, it times out. The flashing tool is not patient enough. I'll try to tweak it. Would be nice to just get this working, since you do that in the browser - less error prone. Idea: The ESP scans for SSIDs, sends them to the browser, user selects them, enters a password and the ESP stores it for later + connects. @Vsaggiomo what do you think? 
> 
> > Idea/suggestion:
> > Since we are not using the LED/Torch as the illumination source anymore, perhaps we can fully ignore it and use full 4-pin mode for saving files to the SD card and don't touch any rtc-stuff in the configuration? What do you think @ethanjli
> 
> Yes, if you can return the SD card-related pins to only be used for the SD card, by removing any LED-related functionality from any of those pins, that change will be a great improvement in making the system's behavior easier to understand and troubleshoot.
Indeed! I haven't done that yet, since the current code seems to be working now. 
> 
> > Interestingly, it only happens after timestamp 41.
> 
> Are you able to get the crash happening always+only after timestamp 41? i.e. is this a deterministic failure, or is there some non-deterministic behavior in this failure? Also, are you able to reproduce the exact same behavior on multiple different SD cards?
This was not reproducible at all. I made some improvements in the code (e.g. moving anglerfishmode + firstboot settings to the EEPROM). Also formatting the SD card with the official tool https://www.sdcard.org/downloads/formatter/ (@Vsaggiomo this should go to the docs - anglerfishdocs?) helped it seems (not sure if broken sectors are getting restored?).
> 
> > @ethanjli do you know any mechanism that probes sectors on the SD card and skips them if necessary?
> 
> I believe you're looking for wear-leveling functionality: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/wear-levelling.html . It looks they recommend using wear-leveling with a filesystem-specific API such as FATFS, rather than using the raw wear-leveling functions.

Will have a look. Thank you! :)
> If you need to store that one into EEPROM, that's fine. However, I would encourage you to still:
>
> 1. Keep one unified interface for managing the persistence of all settings together, so that it [...]
I tried sticking to the interface and replaced only the location where information are getting stored. The problem I see is: If I save a full Json file, chances are high that something is corrupted (e.g. character missing). If that's the case, the file gets removed (one error log stated: file-size 0..). With the single values stored in EEPROM it's much safer. 

> > the stacktrace from above's error leads to some base64 error.
> 
> I'm very surprised to see that it's a base64 _encoding_ error. Can you provide a permalink to line 908 of main.ino?
Indeed, weird. PErmalink: 

https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/e79de02eecd91927efe1e2c695211f5a17776fbf/main/main.ino#L908 and 
https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/e79de02eecd91927efe1e2c695211f5a17776fbf/main/main.ino#L830

The current firmware that is also available through the web flashing tool works and I can report >840 timelapse images being saved on the sd card now :)

```
Lamp: 0%, pwm = 0
Resetting Camera Sensor Settings...
[ 37586][D][main.ino:871] loadAnglerfishCamSettings(): Capturing dummy frame 0
Mean intensity: 145
[ 37631][D][main.ino:871] loadAnglerfishCamSettings(): Capturing dummy frame 1
Mean intensity: 138
[ 37786][D][main.ino:871] loadAnglerfishCamSettings(): Capturing dummy frame 2
Mean intensity: 145
[ 37885][D][main.ino:871] loadAnglerfishCamSettings(): Capturing dummy frame 3
Mean intensity: 136
[ 37985][D][main.ino:871] loadAnglerfishCamSettings(): Capturing dummy frame 4
Mean intensity: 145
[ 38085][D][app_httpd.cpp:1333] saveImage(): Setting PWM Value
Current: 0%, pwm = 0
Capture Requested for SD card save
Lamp: 255%, pwm = 5672
Lamp: 0%, pwm = 0
Saved /data_May_15_2023_113549_timelapse_image_anglerfish_836texp_500
Lamp: 0%, pwm = 0
Updating /esp32cam-preferences.json
[ 44546][D][storage.cpp:107] writeJsonToSSpiffs(): Opening file for writing
[ 44551][D][storage.cpp:112] writeJsonToSSpiffs(): Serializing JSON to String
[ 44552][D][storage.cpp:118] writeJsonToSSpiffs(): Writing JSON to file
Sleeping for 60 s
```

ethanjli, 2023-05-16:
> I tried sticking to the interface and replaced only the location where information are getting stored. The problem I see is: If I save a full Json file, chances are high that something is corrupted [...]

A workaround this would be to split your values into multiple JSON files. However, if you are getting file corruption at some probabilistic rate, it will happen anyways even if you use multiple files. In that case, you would need to have some sort of error-handling code (e.g. once you write the file, you read it back to ensure it's correct; and if the file fails to save correctly, then you try again to write it; and you use the wear leveling mechanism on your flash memory filesystem). You will also get data corruption in EEPROM, though write errors or data corruption may be at a sufficiently low rate that you could tolerate the rate of failures without implementing error-handling code. I personally would encourage you to implement proper error-handling regardless of whether you use EEPROM or SPIFFS, but it's up to you to decide whether that is worth the work/time needed.

> https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/e79de02eecd91927efe1e2c695211f5a17776fbf/main/main.ino#L908 and [...]

Hmmm, this makes the stack trace even more confusing to me. From looking at the order of function calls printed in the stack trace, I'm starting to wonder if the stack trace is incorrect (e.g. perhaps debugging symbols for the functions are out-of-date compared to the actual memory addresses for the functions), or if you have a code error somewhere which is corrupting memory and messing up the stack trace - the sequence of function calls (StartCamera, initAnglerfish, setup, base64_encode_blockend) does not look at all like a function call stack (e.g. StartCamera is being called by setup rather than initAnglerfish, so it doesn't make sense for initAnglerfish to show up between StartCamera and setup in your stack trace).

Since you presumably made some changes to the code and, as a result, were able to make the firmware to run without encountering that same panic, I have no understanding of what has happened here and I can't really help you further on this issue unless you provide me additional information. It sounds like you consider this bug fixed and you're focusing on other bugs now?