Matchboxscope/Matchboxscope #29: Todo's

beniroquai, 2023-05-24:
A loose list of things we would need to carry out for a proper release

## Experiment

- Timelapse Imaging under water (@Vsaggiomo, @beniroquai )
- Timelapse in Incubator  (@ranranking has cells? ;) )

# Mechanics

- Add Magnets 
- Darkfield with LED 

# Documentation

- Add Motorwiring for ESPlanktoscpe 
- Freeze CAD and add new zipped release

# Software 

- test inline hologram reconstruction 
- add XIAO to uploader

arteys, 2023-05-24:
Hi. I apologise if I'm meddling where I shouldn't be, I have no experience of taking part in open projects. 

However, I do have cell cultures and all sorts of histological sections to consider and I can somehow get involved in all of this.

beniroquai, 2023-05-25:
Perfect! Would you be up for some crazy timelapse image series in the incubator? :D 
The wilder it gets the better :D

Vsaggiomo, 2023-05-26:
> A loose list of things we would need to carry out for a proper release
> 
> ## Experiment
> * Timelapse Imaging under water (@Vsaggiomo, @beniroquai ) 
hopefully running right now....
> * Timelapse in Incubator  (@ranranking has cells? ;) )
> 
> # Mechanics
> * Add Magnets
> * Darkfield with LED
* XIAO ? 

* version without inserts
* possibly side-mounted illumination arm (external LED) for using the whole stage
* if possible, unified design (3D printable + injection molding)

> # Documentation
> * Add Motorwiring for ESPlanktoscpe
> * Freeze CAD and add new zipped release
* clean up repository 
* clean up documentation injection molded
* images on the FLASH page, better instructions on which firmware to use and why
* software instructions
* .io remove the stuff not in use, not documented right now. better structures of the folders
* .io keep the matchboxscope as the cheapest possible version / v.0 
 
> # Software
> * test inline hologram reconstruction
> * add XIAO to the uploader
* test the XIAO in the field