Matchboxscope/Matchboxscope #30: NOT an issue just a general question.

lohtse, 2023-07-26:
Love this project.

But wondering which is the best program to load on to the esp32 mod??/ Have everything printed and assembled etc, just want to make sure I have the most uptodate and functional software flashed on...

thank you for an awesome project..

arteys, 2023-07-26:
Hi. You can use online uploader: https://matchboxscope.github.io/firmware/FLASH.html

lohtse, 2023-07-26:
Thats pretty darn helpful!!

Nice new features too camera server too.

Hows does the Timelapse work? Does it just save the pics to device(phone)? if so where?

arteys, 2023-07-26:
No, it saves the images to the sd card

beniroquai, 2023-07-26:
Thanks @arteys for jumping in! 
Good news, I'm developing an app that stores images on the phone too ;-) 
here it is: https://github.com/Matchboxscope/Anglerfish-APP
I'm mostly developing with the XIAO Sense from Seeed Studio these days. Much more reliable in terms of image quality and power outage. @Vsaggiomo knows what I'm talking about. 

The APP mostly helps us to record video data from the camera. It's a work in progress. If you want to see features, please feel free to post an Issue! !:)

lohtse, 2023-07-26:
> 

Thank you for respnding and confirming

lohtse, 2023-07-26:
@beniroquai  

could not get it to install on any of my devices... have left comment on apps own github

beniroquai, 2023-07-26:
What's the error message or behaviour? Were you able to bring it to boot
mode?

lohtse, 2023-07-26:
only error message I got was "could not install"     if I use pc apkinstaller to install it says "missing file" as error

lohtse, 2023-07-27:
Where can I find the STLS for the parts seen here? https://matchboxscope.github.io/docs/ESPressoScope/#improvements

they look awesome and seem much better than the ones I printed form main github..

beniroquai, 2023-07-28:
The APK is still in experimental mode. It's not signed, so maybe you will run into issues when installing it. Did you try entering the developer mode (please google) 

As for the second point, I have updated the release notes in the github.io that should point to the release files here: https://github.com/Matchboxscope/Matchboxscope/releases/tag/v2 

I hope that helps :)

lohtse, 2023-07-28:
all but one device is rooted and all devices have dev mode enabed and same issue on all....

thank you for the STLs will be printing tomorrow...

beniroquai, 2023-07-28:
Interesting. Can you access logcats?

> Message ID: ***@***.***>
>

lohtse, 2023-07-28:
I can but pther than normal stuff literally shows nothing about the failed install

lohtse, 2023-07-28:
this isthe section of the logcat    for time just before, during and after istallg and failed

7-28 22:44:33.797  1739  2595 D GameManagerService:   notifyResumePause(). unexpected mPrevNotiType: -1
07-28 22:44:33.802  1044  1967 D AudioFlinger: mixer(0xef4039c0) Spend too much time to write: delta 110(effect 0, stage 0)
07-28 22:44:33.805  1739  6448 D GamePkgDataHelper: getGamePkgDataIncServer(). com.speedsoftware.rootexplorer
07-28 22:44:33.805  1739  3847 D Dex2oatAdjuster: Before bg size = 0
07-28 22:44:33.805  1739  3847 D Dex2oatAdjuster: After bg size = 0 fg size = 0
07-28 22:44:33.814  3057  3083 D ForegroundUtils: could not check pending caller
07-28 22:44:33.820  1739  6448 D GameManagerService: identifyGamePackage. com.speedsoftware.rootexplorer
07-28 22:44:33.820  1739  6448 D GamePkgDataHelper: getGamePkgData(). com.speedsoftware.rootexplorer
07-28 22:44:33.821  1739  6448 D GameManagerService: identifyGamePackage. com.speedsoftware.rootexplorer
07-28 22:44:33.821  1739  6448 D GamePkgDataHelper: getGamePkgData(). com.speedsoftware.rootexplorer
07-28 22:44:33.822  1739  6448 D GameManagerService: identifyGamePackage. com.speedsoftware.rootexplorer
07-28 22:44:33.822  1739  6448 D GamePkgDataHelper: getGamePkgData(). com.speedsoftware.rootexplorer
07-28 22:44:33.828  1739  2798 D ActivityManager: post active user change for 0 fullscreen true isHomeActivity() false
07-28 22:44:33.830  1739  2798 V WindowManager: Relayout Window{2892827 u0 com.speedsoftware.rootexplorer/com.speedsoftware.rootexplorer.RootExplorer}: viewVisibility=0 req=1080x1920 WM.LayoutParams{(0,0)(fillxfill) sim=#10 ty=1 fl=#81810100 pfl=0x20000 wanim=0x10302fd vsysui=0x700 needsMenuKey=2 colorMode=0 naviIconColor=0}
07-28 22:44:33.832   756   756 I SurfaceFlinger: id=907 createSurf (1080x1920),1 flag=404, com.speedsoftware.rootexplorer/com.speedsoftware.rootexplorer.RootExplorer#0
07-28 22:44:33.836  1739  2798 V Surface : sf_framedrop debug : 0x4f4c, game : false, logging : 0
07-28 22:44:33.836  1739  1960 D WindowManager: set systemUiVisibility : systemUiFlags= 0x708 fullscreenStackSysUiFlags= 0x0
07-28 22:44:33.840  1044  2560 V APM_AudioPolicyManager: getAudioPolicyConfig: audioParam;outDevice
07-28 22:44:33.840  1044  2560 V APM_AudioPolicyManager: getNewOutputDevice() selected device 0
07-28 22:44:33.841  1044  2560 V APM_AudioPolicyManager: ### curdevice : 2
07-28 22:44:33.842  1739  2798 D InputDispatcher: Focus entered window: 11827
07-28 22:44:33.845 11827 11827 V Surface : sf_framedrop debug : 0x4f4c, game : false, logging : 0
07-28 22:44:33.845 11827 11827 D ViewRootImpl@9f650f4[RootExplorer]: Relayout returned: old=[0,0][1080,1920] new=[0,0][1080,1920] result=0x7 surface={valid=true 512888291328} changed=true
07-28 22:44:33.848  1739  5001 D WindowManager: adjustSystemUiVisibilityLw : vis= 0x708
07-28 22:44:33.850  1044  2560 V APM_AudioPolicyManager: getAudioPolicyConfig: audioParam;outDevice
07-28 22:44:33.850  1044  2560 V APM_AudioPolicyManager: getNewOutputDevice() selected device 0
07-28 22:44:33.850  1044  2560 V APM_AudioPolicyManager: ### curdevice : 2
07-28 22:44:33.852  1044  2560 V APM_AudioPolicyManager: getAudioPolicyConfig: audioParam;outDevice
07-28 22:44:33.852  1044  2560 V APM_AudioPolicyManager: getNewOutputDevice() selected device 0
07-28 22:44:33.852  1044  2560 V APM_AudioPolicyManager: ### curdevice : 2
07-28 22:44:33.859  1044  1315 V APM_AudioPolicyManager: stopOutput() output 21, stream 1, session 65
07-28 22:44:33.859  1044  1315 V APM_AudioPolicyManager: setBeaconMute(1) mBeaconMuteRefCount=0 mBeaconPlayingRefCount=0
07-28 22:44:33.860  1044  1315 V APM_AudioPolicyManager: getNewOutputDevice() selected device 0
07-28 22:44:33.860  1044  1315 V APM_AudioPolicyManager: setOutputDevice() device 0000 delayMs 96
07-28 22:44:33.860  1044  1315 V APM_AudioPolicyManager: setOutputDevice() prevDevice 0x0002
07-28 22:44:33.860  1044  1315 V APM_AudioPolicyManager: setOutputDevice() setting same device 0x0000 or null device
07-28 22:44:33.862 11827 11868 D OpenGLRenderer: eglCreateWindowSurface = 0x777a614980
07-28 22:44:33.887  1044  2560 V APM_AudioPolicyManager: getAudioPolicyConfig: audioParam;activeStream
07-28 22:44:33.887  1044  2560 V APM_AudioPolicyManager: ### active stream : 0
07-28 22:44:33.887  1739  6448 D AudioService: active stream is 0x0
07-28 22:44:33.891  1739  6448 D GameManagerService: identifyGamePackage. com.speedsoftware.rootexplorer
07-28 22:44:33.891  1739  6448 D GamePkgDataHelper: getGamePkgData(). com.speedsoftware.rootexplorer
07-28 22:44:33.892  1739  6448 D GameManagerService: identifyGamePackage. com.speedsoftware.rootexplorer
07-28 22:44:33.892  1739  6448 D GamePkgDataHelper: getGamePkgData(). com.speedsoftware.rootexplorer
07-28 22:44:33.899  1739  5001 V WindowManager: finishDrawingLocked: mDrawState=COMMIT_DRAW_PENDING Window{2892827 u0 com.speedsoftware.rootexplorer/com.speedsoftware.rootexplorer.RootExplorer} in Surface(name=com.speedsoftware.rootexplorer/com.speedsoftware.rootexplorer.RootExplorer)
07-28 22:44:33.902 11827 11827 D ViewRootImpl@9f650f4[RootExplorer]: MSG_WINDOW_FOCUS_CHANGED 1
07-28 22:44:33.912  1739  6448 D SSRM:Q  : SIOP:: AP = 360, PST = 348 (W:22), BAT = 296, CHG = 344
07-28 22:44:33.919 11827 11827 V InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@c19a09b nm : com.speedsoftware.rootexplorer ic=null
[... cut ...]
07-28 22:44:34.162  1739  2798 W ActivityManager: No permission grants found for com.degoo.android
07-28 22:44:34.246  1739  2798 W ActivityManager: No permission grants found for com.degoo.android
07-28 22:44:34.332  1739  6448 D SSRM:Q  : SIOP:: AP = 360, PST = 350 (W:24), BAT = 296, CHG = 344
07-28 22:44:34.345  1739  2798 W ActivityManager: No permission grants found for com.degoo.android
07-28 22:44:34.451 28490 28502 I zygote64: Background concurrent copying GC freed 135452(8MB) AllocSpace objects, 2(40KB) LOS objects, 49% free, 8MB/16MB, paused 1.363ms total 117.256ms

lohtse, 2023-07-28:
+----------------------------+---------------------+------
07-28 22:52:04.342   756   756 I SurfaceFlinger:      Device | 0x70d32c0680 | 0002 | RGBA_8888 |   0.0    0.0 1080.0 1920.0 |    0    0 1080 1920 | com.google.android.packageinstaller/[...]installer.PackageInstallerActivity#0
07-28 22:52:04.342   756   756 I SurfaceFlinger:      Device | 0x70d0d7bc00 | 0000 | RGBA_8888 |   0.0    0.0 1080.0   63.0 |    0    0 1080   63 | StatusBar#0
07-28 22:52:04.342   756   756 I SurfaceFlinger:      Device | 0x70cdbec700 | 0000 | RGBA_8888 |   0.0    0.0   87.0   87.0 |  947 1810 1034 1897 | Sprite#2
07-28 22:52:04.342   756   756 I SurfaceFlinger: 
07-28 22:52:04.384  1739  2430 D InputReader: Input event(5): value=0 when=38625125186000
07-28 22:52:04.384  1739  2430 D InputReader: Input event(5): value=0 when=38625125186000
07-28 22:52:04.384  1739  2430 I InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=38625125186000
07-28 22:52:04.385  1739  1739 D InputDispatcher: Inject motion (0): action=0x1, eventTime=38625125186000, displayId=0
07-28 22:52:04.385  1739  2429 I InputDispatcher: Delivering touch to (12045): action: 0x1, toolType: 1
07-28 22:52:04.386 12045 12045 D ViewRootImpl@37ba3e8[PackageInstallerActivity]: ViewPostIme pointer 1
07-28 22:52:04.390  1739  2493 V AudioService: AudioService playSoundEffect Type : 0, volume : -1.0
07-28 22:52:04.391  1739  2550 V AudioService: received MSG_PLAY_SOUND_EFFECT onPlaySoundEffect start
07-28 22:52:04.391  1044  1358 V APM_AudioPolicyManager: getAudioPolicyConfig: situation=1;device=0
07-28 22:52:04.391  1044  1358 I APM_AudioPolicyManager: getSituationVolume: isBTConnected:0 isHeadConnected:0
07-28 22:52:04.391  1044  1358 V APM_AudioPolicyManager: deviceInt = 0, situationInt = 1, situationVolume = 0.500000
07-28 22:52:04.392  1044  1358 V APM_AudioPolicyManager: getOutput() device 2, stream 1, samplingRate 0, format 0, channelMask 3, flags 0
07-28 22:52:04.392  1044  1358 V APM_AudioPolicyManager: selectOutput() commonFlags for output 13, 0000
07-28 22:52:04.392  1044  1358 V APM_AudioPolicyManager: selectOutput() commonFlags for output 21, 0000
07-28 22:52:04.392  1044  1358 V APM_AudioPolicyManager: selectOutput() commonFlags for output 29, 0000
07-28 22:52:04.392  1044  1358 V APM_AudioPolicyManager:   getOutputForDevice() returns output 13
07-28 22:52:04.392  1044  1358 V APM_AudioPolicyManager: getOutput() device 2, stream 1, samplingRate 0, format 0, channelMask 3, flags 0
07-28 22:52:04.392  1044  1358 V APM_AudioPolicyManager: selectOutput() commonFlags for output 13, 0000
07-28 22:52:04.392  1044  1358 V APM_AudioPolicyManager: selectOutput() commonFlags for output 21, 0000
07-28 22:52:04.392  1044  1358 V APM_AudioPolicyManager: selectOutput() commonFlags for output 29, 0000
07-28 22:52:04.393  1044  1358 V APM_AudioPolicyManager:   getOutputForDevice() returns output 13
07-28 22:52:04.385  1739  1739 D AccessibilityInputFilter: mSamsungMagnifierWindow : null
07-28 22:52:04.393  1739  6448 D GameManagerService: identifyGamePackage. com.google.android.packageinstaller
07-28 22:52:04.393  1739  6448 D GamePkgDataHelper: getGamePkgData(). com.google.android.packageinstaller
07-28 22:52:04.394  1044  1358 W AudioPolicyIntefaceImpl: Skipped to add effects on session 65
07-28 22:52:04.394  1044  1358 V APM_AudioPolicyManager: startOutput() output 21, stream 1, session 65
07-28 22:52:04.394  1044  1358 V APM_AudioPolicyManager: setBeaconMute(1) mBeaconMuteRefCount=1 mBeaconPlayingRefCount=0
07-28 22:52:04.394  1044  1358 V APM_AudioPolicyManager: getNewOutputDevice() selected device 2
07-28 22:52:04.394  1044  1358 V APM_AudioPolicyManager: setOutputDevice() device 0002 delayMs 0
07-28 22:52:04.394  1044  1358 V APM_AudioPolicyManager: setOutputDevice() prevDevice 0x0002
07-28 22:52:04.394  1044  1358 V APM_AudioPolicyManager: setOutputDevice() setting same device 0x0002 or null device
07-28 22:52:04.394  1044  1358 I AudioFlinger: updateStageIndex prev 0, new -2101
07-28 22:52:04.397  1739  4019 I ActivityManager: START u0 {act=null typ=null flg=0x2000000 cmp=ComponentInfo{com.google.android.packageinstaller/com.android.packageinstaller.InstallInstalling}} from uid 10027
07-28 22:52:04.399   728   793 E ANDR-PERF-MPCTL: Invalid profile no. 0, total profiles 0 only
07-28 22:52:04.399  1739  4019 D CustomFrequencyManagerService: acquireDVFSLockLocked : type : DVFS_MIN_LIMIT  frequency : 1632000  uid : 1000  pid : 1739  pkgName : AMS_ACT_START@CPU_MIN@83
07-28 22:52:04.400  1739  4019 D ActivityManagerPerformance: AMP_acquire() ACT_START
07-28 22:52:04.418  1739  5001 D ActivityManager: isScaleDownAnimationEnabled() : false
07-28 22:52:04.422  1739  5001 D PersonaManagerService: Current focused persona service handled id set to : 0
07-28 22:52:04.424  1739  5001 D InputDispatcher: Focused application set to: xxxx
07-28 22:52:04.425  1739  5001 D InputDispatcher: Focus left window: 12045
07-28 22:52:04.428  1739  2595 D GameManagerService: handleForegroundChange(). pkgName: com.google.android.packageinstaller, clsName: com.android.packageinstaller.InstallInstalling,FgActivityName:com.google.android.packageinstaller/com.android.packageinstaller.InstallInstalling
07-28 22:52:04.428  1739  2595 D GameManagerService:   handleForegroundChange(). same package. game has never resumed yet. ignore
07-28 22:52:04.429  1739  5001 D MARsPolicyManager: onPackageResumedFG pkgName = com.google.android.packageinstaller, userId = 0
[... cut ...]
07-28 22:52:08.200 12045 12045 D InputEventReceiver: channel '55e4d1e com.google.android.packageinstaller/com.android.packageinstaller.InstallFailed (client)' ~ Disposing input event receiver.
07-28 22:52:08.200 12045 12045 D InputEventReceiver: channel '55e4d1e com.google.android.packageinstaller/com.android.packageinstaller.InstallFailed (client)' ~NativeInputEventReceiver.
07-28 22:52:08.326  1739  2438 D MdnieScenarioControlService:  packageName : com.speedsoftware.rootexplorer    className : com.speedsoftware.rootexplorer.RootExplorer

lohtse, 2023-07-29:
Another question about the stl models,   what is the actual external LED you have used???

Vsaggiomo, 2023-07-29:
it's in the UPDATE of https://matchboxscope.github.io/docs/Matchboxscope 
They are LIHAO mini LED (https://www.amazon.nl/LIHAO-mini-ledlampjes-lantaarns-batterijen-waterdicht/dp/B01FF07XFE )

beniroquai, 2023-08-02:
@lohtse, It's a bit hard for me to debug the app given the logs. What was the behaviour? Could you install android studio and compile it yourself to see what may cause the issue by any chance?

lohtse, 2023-08-02:
Hi buddy

already explained what happens... that is litereally it...  

As to Android studio... Will make no difference too me as I will have no clue as to what is what etc... Hard enough, working with the various platforms for esp32/ard  devices, Same with the radios am involved in modding. and even with Garmin CIQ!!  have alittle too much on the plate too start with sometjing else.

sorry about that buddy

beniroquai, 2023-08-07:
Hey @lohtse, fixed some issues and made a tutorial. Does this work for you? 
https://matchboxscope.github.io/docs/APP

beniroquai, 2023-08-07:
True. Just an attempt to record videos. You could speed up streaming if you reduce the resolution I guess. 
If you can speak Android, I'm happy to work on PR ;-) 
Perhaps @KillerInk knows more about it :-P

lohtse, 2023-08-07:
> True. Just an attempt to record videos. You could speed up streaming if you reduce the resolution I guess.
If you can speak Android, I'm happy to work on PR ;-)
Perhaps @KillerInk knows more about it :-P

Sadly my android is a little lacking

for me changing res made little difference. where as the browser even at high res workeds fine..

But as you have said this is still in dev so at least there is a start and it can only get better ;-)   thank you for great work. I do appreciate it..

Kids love the microscope and we have now built 5 for the school too for their field trips(at the schools resquest)!

KillerInk, 2023-08-08:
> True. Just an attempt to record videos. You could speed up streaming if you reduce the resolution I guess. If you can speak Android, I'm happy to work on PR ;-) Perhaps @KillerInk knows more about it :-P

the problem is on app side =P
you should implement a framebuffer and a seperate drawing thread.
also drawing into a imageview is not the best idea. but for now its ok.

KillerInk, 2023-08-08:
i played around a bit. and on higher resolution speed went down in the app and browser.
but the browser have one option that the app dont have. XCLK 
increasing that value increase also the frame rate due faster clock^^
on that esp32cam board its by default at 8mhz. but could easly set to 20mhz

beniroquai, 2023-08-08:
Until breaks :D 
The ESP32Cam is  - well - shitty. 
@Vsaggiomo had this picture of "reproducibility" tests. Same hardware, but completely different results. ;)

lohtse, 2023-08-09:
> Until breaks :D The ESP32Cam is - well - shitty. @Vsaggiomo had this picture of "reproducibility" tests. Same hardware, but completely different results. ;)

For the price point the esp32cam is not bad and pne you don't mind killing, but if you can suggest a better one for the size and price then please do...

KillerInk, 2023-08-09:
Esp cam is fine with that sensors age.
That clock rate is close to exposure time, it affects the readout time. And gain can be seen as iso.
At least both have something todo with the math that get used by ae.