Matchboxscope/Matchboxscope #31: Board does not show up on Ubuntu 22.04 Arduino IDE 2.1.0

SarenT, 2023-09-01:
I wonder if anyone else has the same issue, perhaps the ESP32-CAM MB is not recognized but the board does not appear under Port.

SarenT, 2023-09-01:
It turns out to be a driver (CH340) issue, which after installing [](https://github.com/juliagoda/CH341SER) worked. I was able to upload at least. But it was also more than that. brltty service was claiming this driver and uninstalling it was necessary, which would impact blind people. Perhaps some info can be added to the documentation.

beniroquai, 2023-09-01:
Interesting indeed. I had problems with the webserial on Ubuntu recently. Would you mind noting the steps you walked until it worked?

SarenT, 2023-09-01:
Of course! I realized that the unnamed links do not show up at all. I have just corrected my previous comment.
It was basically installing the driver and uninstalling the `brltty`. After that I had an issue with the device being busy.
I was also checking with `dmesg` to see if the device is recognized properly (`/dev/ttyUSB*`).

beniroquai, 2023-09-01:
Interesting! I thought the driver was already pre-shipped. That's not the case apparently?