Matchboxscope/Matchboxscope #32: Confused about optical theory

zhouruhui, 2023-12-27:
"Matchboxscope.md" shows a optical formula, 1/f= 1/a'-1/a, is that means a'<f? The sample and its virtual image should be in the same side.
It seems to be different with "we have a sample at a distance of 2f (focal lengths) in front of the lens, which will in turn create an image at 2f away from it. "

beniroquai, 2023-12-27:
Hey @zhouruhui thanks for reaching out. Indeed, the magnification heavily depends on the distances between sample, lens and lens, sensor. I would not put it in stone as of now, since people may change e.g. the adapter to change the magnification. The 2f system (or 4f in total) states that you have a 1:-1 magnification. Does this make sense to you?

zhouruhui, 2023-12-30:
@beniroquai Got it. Thanks for your reply!

beniroquai, 2023-12-30:
My Pleasure :)

closed 2023-12-30 by beniroquai